ra3xdh/qucs_s #826: Improve converter tool GUI

ra3xdh, 2024-07-08:
I propose the following improvements for converter  tool GUI (accessible from *Tools->Data files converter*)

- [ ] Add possibility to define the library name: https://github.com/ra3xdh/qucsator_rf/issues/11
- [ ]  Add possibility to select data variable from drop-down list instead type-in manually

tomhajjar, 2024-07-08:
Presently to create/repair/update a library with the same name as an existing one, the existing one has to be deleted/renamed.

Will Qucs-S now allow overwriting libraries in usr_lib? 

Will Qucs-S now allow creating libraries in usr_lib that have the same name as a library in C:\QUCS-S 24.2.0\share\qucs-s\library?

Th reason I ask is the library filename is also embedded throughout the file. Renaming the library doesn't work. You have to replace every filename instance throughout the library.

ra3xdh, 2024-07-13:
The proposed improvements implemented by #841

closed 2024-07-13 by ra3xdh