ra3xdh/qucs_s #846: Library extension

ra3xdh, 2024-07-17:
I have recently discovered this website containing SPICE models for different devices: https://fotoelektronika.com/spice-models/ The most of these models could be converted to Qucs-S libraries. The models requires porting to Ngspice notation (replace `IF` by ternary operator etc.).

ra3xdh, 2024-07-18:
I have tested the provided PWM controllers models, and it doesn't work with Ngspice. The Ngspice reads the model without error but then fails to converge or gives kilovolt outputs. Theses models requires significant effort to port them to Ngspice syntax.

tomhajjar, 2024-07-18:
Can you post your project? The ngspice guys have made many changes to improve compatibility. They also have recommended changes to models.

ra3xdh, 2024-07-18:
Here is the project. It is done using the latest snapshot build. The MC34063 device is taken from another place and works as expected. The both UC3483 and TL494 gives either zero or kilovolt output. I suppose the problem may come from the following line in the models:

~~~
.IC V(QINT,GND) 0
~~~ 

I have tried replacing this by setting IC on the capacitor connected to this node, but no result. I also see nested `IF` conditions in one source. I am not sure if NGspice supports this.

~~~
GQ  GND  QINT VALUE {IF(V(CLKI,GND)>{V(VCC,GND)/2},IF(V(D,GND)>{V(VCC,GND)/2},{V(VCC,GND)},-5),0)}
~~~

[PWM_controller_prj.zip](https://github.com/user-attachments/files/16285056/PWM_controller_prj.zip)

ra3xdh, 2024-07-19:
I finally managed to get TL494 operational. The .IC directive required correction. Also setting `InitialDC=no` helps. The UC3843A model still be no result. 

[image]

[PWM_controller_prj.zip](https://github.com/user-attachments/files/16310953/PWM_controller_prj.zip)

ra3xdh, 2024-07-19:
I have added MixerIC library containing SA612 model as part of #850. Everything works as expected with default Ngspice settings.

[image]

tomhajjar, 2024-07-19:
I will make a symbol for the Tl494.

I worked on the SA612 many years ago. Attached is my project. I made two different SA612 symbols but the colors and line widths don't conform to what we are using today.

[image]
[image]
[image]

[SA612_prj.zip](https://github.com/user-attachments/files/16311441/SA612_prj.zip)

ra3xdh, 2024-07-19:
I have fixed the UC3843A model. There was a mistake in testbench schematic in PDF. The supply voltage should be in range from 9 to 15V. Also the DX diode model required a correction. The default BV value caused convergence error for Ngspice. The fixed model is attached. The other UC384x devices may be fixed in a similar way.

[image]

[UC3843A.cir.txt](https://github.com/user-attachments/files/16313379/UC3843A.cir.txt)

ra3xdh, 2024-07-19:
Summarizing all above, the PWM controller models could be assembled in a library. This library could be added in the upcoming release.

ra3xdh, 2024-07-19:
I have added TODO list to this issue.

tomhajjar, 2024-07-19:
First cut of the TL494 symbol. Since the device can be used to make multiple regulator topologies, no obvious way to make the symbol.

[image]

[TL494.zip](https://github.com/user-attachments/files/16316364/TL494.zip)

ra3xdh, 2024-07-22:
PWM controller library implemented by #855.

tomhajjar, 2024-07-22:
I made a small change to the TL494 symbol. PIN3, CMP->FBK

CMP was on an old schematic. FBK is on data sheet...

[image]

[TL494.zip](https://github.com/user-attachments/files/16335744/TL494.zip)

tomhajjar, 2024-07-23:
There is a Kicad example using the UC1825. I haven't tested it.

[image]

[UC1825.zip](https://github.com/user-attachments/files/16343234/UC1825.zip)

tomhajjar, 2024-07-23:
SG3525A is very similar and made by many companies. I'm looking into it.

tomhajjar, 2024-09-03:
I have updated the Avago ACPLK30T Photo Voltaic Isolator. It would make a good addition to Optocoupler.lib

[image]
[image]

[ACPLK30T_Photo_Voltaic_Isolator.zip](https://github.com/user-attachments/files/16858786/ACPLK30T_Photo_Voltaic_Isolator.zip)

closed 2025-01-22 by ra3xdh