ra3xdh/qucs_s #856: Location specific user library references

KrzysztofHerman, 2024-07-22:
In our  [IHP-Open-PDK](https://github.com/IHP-GmbH/IHP-Open-PDK) we provide both: (1) device primitives symbols and (2) schematic/simulation examples which allow the user to design circuits using Qucs-S and `IHP-Open-PDK`. The symbol library is placed under `.qucs/user_lib/` directory in user's home location.  Unfortunately the examples we provide use fixed references of the symbols as follows:
```
  <Lib dpantenna1 1 470 750 40 -16 0 0 "/home/herman/.qucs/user_lib/IHP_PDK_nonlinear_components" 0 "dpantenna" 0 "16u" 1 "16u" 1>
  <Lib dantenna1 1 270 750 40 -16 0 0 "/home/herman/.qucs/user_lib/IHP_PDK_nonlinear_components" 0 "dantenna" 0 "8u" 1 "8u" 1>
```
what, by default does not pull the symbols from the `.qucs/user_lib/` because of the specific username. 

Is there any other way to distribute the symbols and examples ensuring portability at least for linux users ?

More information about qucs-s and IHP-Open-PDK can be found [here](https://github.com/IHP-GmbH/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.tech/qucs)

ra3xdh, 2024-07-22:
Qucs-S should support relative paths in libraries since the upcoming  version v24.3.0. Furthermore, it shows the libraries from the project directory in the project tree. See #567 for more details.

ra3xdh, 2024-09-02:
No further explanation provided. Assume #567 resolves the mentioned issues. Closing.

closed 2024-09-02 by ra3xdh