ra3xdh/qucs_s #861: Spice model with "+" for additional input PINs

tomhajjar, 2024-07-23:
Ran into an odd issue. It seems ngspice can handle a Spice model with "+" for additional input PINs but Qucs-S cannot.

.SUBCKT UC1825_TRANS OUT_B OUT_A ILIM-SD RAMP CT GND Vcc VC VREF NI INV E-A_OUT CLOCK 
+ RT GND SS 

[UC1825_TRANS_ng.zip](https://github.com/user-attachments/files/16350316/UC1825_TRANS_ng.zip)

ra3xdh, 2024-07-23:
Line continuation in the `.SUBCKT` header is not supported. The same is valid for old SpiceFile device. Edit the SPICE model to combine the lines as the workaround.

tomhajjar, 2024-07-23:
I only mentioned it because of KiCad/ngspice. 

Another issue is the model has two pins called GND instead of GND and PGND yet keeps VCC and VVC separate.

I am waiting for a response on the KiCad forum why the simulation fails. Doing simulations in KiCad are frustrating because old simulations are not guaranteed to work in new versions of KiCad. Been burned many times...

ra3xdh, 2024-11-18:
Fixed by #1074

closed 2024-11-18 by ra3xdh