ra3xdh/qucs_s #863: Component windows items looks blury version 24.3

dsm, 2024-07-24:
Hi in new release 24.3 items looks blury but nothing blury in schematics , you can look difference in above image.

[image]

wawuwo, 2024-07-25:
@dsm could you please build with [this](https://github.com/ra3xdh/qucs_s/blob/dedcd59bf4b5e60801484d24703b747149d76230/qucs/components/component.cpp#L326) line commented out and check if it makes any difference in icons appearance?

dsm, 2024-07-25:
Hi, I'm disabling 326 and 327 both or not nothing changed so I changed renderhints nothing changed and give worst results.

wawuwo, 2024-07-26:
Well, I ran out of ideas for now 🤔

ra3xdh, 2024-07-26:
I noticed some blur, but I thought it was Wayland issue. 

> Well, I ran out of ideas for now

If it is not clear how fix, the things may be kept as is.

wawuwo, 2024-08-02:
There are some devices/elements which use SVG-images as their icons. Do they look blurry too? For example, simulations or paintings

dsm, 2024-08-02:
there is no diff between simulation icons looks same.

ra3xdh, 2024-08-26:
I believe the things could be kept as is and this issue may be closed.

closed 2024-08-26 by ra3xdh