ra3xdh/qucs_s #864: Qucs attenuator 24.3.0 gets lost in i18n environment

csrabak, 2024-07-24:
In presente Qucs-s release, in the Windows edition, running Windows 10, Brazilian Portuguese version (single language), Qucs Attenuator doesn't allow me to enter a decimal value of Pin in mW.

If I type 62,5 [mW] (chosen from the drop list at right) just as the focus of this entry field is left, the tool puts forth a value of 625,0!

If instead I decide to enter value in W the tool initially doesn't let putting more than two decimal places, however, as the focus leaves the tool bring the value from 0,06 to 6 W‼

My belief is something related to internationalization stems from the fact that the dissipated values calculated are shown with dot as decimal separator.

I've no means of showing this short of sending a kind of movie... 

TIA

ra3xdh, 2024-07-24:
The comma is not a valid decimal delimiter for Qt applications. This is limitation of Qt library. Use dot '.' as the decimal delimiter.

andresmmera, 2024-07-25:
I can reproduce the problem.
@ra3xdh I think I can provide a fix soon for this.

closed 2024-07-26 by ra3xdh