ra3xdh/qucs_s #869: Active Filter Synthesis Tool Won't Launch (windows)

BuckyAH, 2024-07-26:
The active filter synthesis tool won't launch but all other tools will. No errors are given when attempting to launch the tool, nothing happens at all. (I am on the windows version if this matters)

dsm, 2024-07-26:
I think Qt6SvgWidgets.dll is missing.

BuckyAH, 2024-07-26:
> I think Qt6SvgWidgets.dll is missing.

I originally had an error saying this and it said to reinstall. When I did this I no longer had the error and had the current error of nothing happening at all. Do you think its still missing but just not giving the error message? If so, how could I fix this?

dsm, 2024-07-26:
Installer will be reuploaded with missing .dll but you can download dll and put the bin folder of the program. 

[Qt6SvgWidgets.zip](https://github.com/user-attachments/files/16391175/Qt6SvgWidgets.zip)

ra3xdh, 2024-07-26:
I will reupload the installer soon.

ra3xdh, 2024-07-26:
I have added the missing DLL in the installer. Closing as completed.

closed 2024-07-26 by ra3xdh