ra3xdh/qucs_s #871: add CI support for stable release.

dsm, 2024-07-26:
Hi, I'am planning to improve our CI to make release more consistent. I added below todo list for this issue. If any steps needs that I missed, please tell me.

- [ ] add windows installer support our CI.
- [ ] add misc folder to repo.
- [ ] add inno setup file to repo.
- [ ] add ngspice to windows release.
- [ ] CI should be automatically run for newly created tag and make release.
- [ ] check version and the tag name matching.
- [ ] insert automatic changelog comparing previous tag to release for maybe NEWS.md not ready.

dsm, 2024-08-22:
@ra3xdh can you add misc folder and inno setup file to repo ?

ra3xdh, 2024-08-22:
See #901 I have uploaded install scripts in `contrib` subdirectory.

dsm, 2024-08-24:
@ra3xdh Hi, you build deb and rpm using obs we can add this to our CI to your help, there are github actions for this and runs on ubuntu I added links for them. also we can run freebsd using this action.
https://github.com/marketplace/actions/freebsd-vm

https://github.com/marketplace/actions/build-debian-packages
https://github.com/marketplace/actions/rpm-build

ra3xdh, 2024-08-25:
It's need to learn the features of Github Actions approach. OBS provides not only DEB/RPM packages build, but also repository infrastructure and automatic generation of installation instructions. Also OBS allows to build the packages for a number of distribution versions. I am not inclined to switch the DEB/RPM generation to Github actions.

zergud, 2024-08-26:
I suggest changing the versioning for CI, may be 24.3.99
same version as Release may be confusing
[image]

dsm, 2024-08-26:
@zergud yes, I can do this, it's easy. 24.3.99 or next release ex. 24.4.0 maybe and maybe added git hash. @ra3xdh what is your opinion ?

ra3xdh, 2024-08-26:
Adding the `99` at the end of continuous release may have side effects. The application checks all three digits in the libraries and schematic. And the schematic or library created using continuous release will show warning in the next stable release for example if releasing `v24.3.1`. It's need to modify the document version check first. Probably comparing only major version would be sufficient.

dsm, 2024-08-26:
Hi, @ra3xdh I'll send a PR for this, I set nightly version to 23.4.99 and normal release use tag version and hide git hash if it's stable release. I don't use VERSION file for this just used local builds. Now version set is flexible.

dsm, 2024-08-29:
I am done with this issue.

ra3xdh, 2024-08-29:
Closed as completed. Thank you @dsm !

closed 2024-08-29 by ra3xdh