ra3xdh/qucs_s #872: Ubuntu 22.04 updater fail due to file size mismatch

PaulKrinsky, 2024-07-26:
Вывод "sudo apt-get upgrade"
Get:1 http://ftp.lysator.liu.se/pub/opensuse/repositories/home%3A/ra3xdh/xUbuntu_22.04  qucs-s 24.3.0-1 [4 042 kB]
Err:1 http://ftp.lysator.liu.se/pub/opensuse/repositories/home%3A/ra3xdh/xUbuntu_22.04  qucs-s 24.3.0-1
  File has unexpected size (4042510 != 4042452). Mirror sync in progress? [IP: 130.236.254.251 80]
  Hashes of expected file:
   - SHA256:c00ecd82f9aa4e20faf60dcc956e648c3dc3b1d0c332c30009d8bb9e9edabd24
   - SHA1:71c9e3bceed2c7c55925ba665f47a6f87cd39a18 [weak]
   - MD5Sum:ee54a1c21c7f288ffe5c373a7eb1d7d7 [weak]
   - Filesize:4042452 [weak]

Размер реального файла не совпадает с размером из "Packages", в итоге обновление обламывается и вообще ничего не ставится уже несколько дней.

ra3xdh, 2024-07-26:
Удалил репозитории и создал заново, но ошибка осталась. Почему так происходит непонятно. Этот размер файлов нигде вручную не прописывается, и OBS должен проставлять его автоматически. Если не получается подключить репозиторий, то следует воспользоваться ссылков "Grab binary package" и установить DEB пакет через `dpkg`. Здесь нужна помощь от специалистов по OBS почему так происходит.

ra3xdh, 2024-07-26:
I have removed the DEB repositories and created it again, but the error still present. I didn't understand what is wrong with OBS. OBS doesn't provide any mean to specify the DEB file size. It should be calculated automatically. The workaround is to install DEB package with `dpkg`.  The help from OBS experts is highly welcome.

ra3xdh, 2024-07-27:
Репозиторий для Дебиана починил. Помогла ручная установка версии `-2`. Видимо старая версия где-то закешировалась на серверах OBS. Сейчас установка на чистой виртуалке работает. 

I have fixed the Debian repository. I suppose the old package version was cached somewhere in the OBS servers. The manual setting extraversion `-2` helped to fix this. The installation in the fresh VM works as expected.

closed 2024-07-27 by ra3xdh