ra3xdh/qucs_s #873: Symbol Text Issue Still Not Fixed

tomhajjar, 2024-07-27:
The text font/size/position used in a subcircuit symbol still does not match the symbol once made into a library. I thought this was addressed before. 

Compare TL494 Symbol in TL494.sch to the one in PWM_Controller.lib

[image]

[image]

[TL494.zip](https://github.com/user-attachments/files/16397558/TL494.zip)

wawuwo, 2024-07-30:
I believe I understand what is the cause of the problem, but I'm not so sure.

When the symbol is saved its elements are saved as paintings (a line, a rect, a text, etc.) and in that case texts have their sizes measured in points (pts). But texts as parts of a symbol have to have their sizes in pixels, so when a symbol is loaded the sizes in points are treated as sizes in pixels, hence the "scaling".

I'll try to come up with a fix.

tomhajjar, 2024-08-22:
Any progress on this? The issue didn't exist in March when I did the attached project. Any symbol with text looks OK when making and editing the symbol but looks bad when the symbol is placed on a schematic.

[image]
[image]

[Digital_Analog_Mixed_prj.zip](https://github.com/user-attachments/files/16713718/Digital_Analog_Mixed_prj.zip)

wawuwo, 2024-08-25:
@tomhajjar hi! Check the linked PR please, it should fix the problem. I've added explicit conversion from points to pixels, when symbol is read by component. But there is drawback with this approach: conversion from points to pixels gives different results on different displays because different pixel density. So it's more like a workaround than a proper fix.

tomhajjar, 2024-08-25:
Will do when the latest Windows nightly is released

ra3xdh, 2024-08-25:
Fixed by #912.

closed 2024-08-25 by ra3xdh

tomhajjar, 2024-08-26:
I checked a number of old projects and the update is a big improvement.

PR #912 (2024-08-25): Fix font size alteration in subcircuit component symbol
Hi! This is an attempt to fix #873