ra3xdh/qucs_s #874: Selection box disappeared in component selection

dsm, 2024-07-28:
Hi, selection box disappeared in component selection.

[selection.webm](https://github.com/user-attachments/assets/eb7425a6-ee1a-45c1-abfe-b24845e8ada5)

ra3xdh, 2024-07-28:
This may be platform-sepcific issue. Noting has changed using Linux+Wayland. 

[Screencast_20240728_164757.webm](https://github.com/user-attachments/assets/88bab2f7-0465-44a5-8e2a-4d165c56a6ad)

dsm, 2024-07-28:
Yes I tested on macOS, there is no issue. Just windows only.

tomhajjar, 2024-07-29:
Not seeing it on Win 10

dsm, 2024-07-29:
Yes I changed this issue title to windows 11

dsm, 2024-07-29:
@ra3xdh I solve this issue to adding this to qucs.cpp@1040 but this may not be best solution.

```c++
#ifdef Q_OS_WIN
  if("11" == QSysInfo::productVersion()){
    for(int sel=0;sel<CompComps->count();sel++){
      CompComps->item(sel)->setBackground(Qt::transparent);
    }
    CompComps->item(i)->setBackground(Qt::lightGray);
  }
#endif
```

this is the result:

[image]

ra3xdh, 2024-07-29:
It seems Windows 11 works wrong with style sheet at this line:
~~~
CompComps->setStyleSheet("QListWidget{background: white; color: black;}");
~~~

Maybe something should be set explicitly in the style sheet if Windows 11 is used.

dsm, 2024-07-29:
I changed white and black background and icon text color changed.

dsm, 2024-07-29:
seems selection-background-color not working.

dsm, 2024-07-29:
I solve this issue using this stylesheet but it has a minor issue icon position changed a bit. @ra3xdh  can you test this stylesheet in linux maybe we can apply all releases.

[selection.webm](https://github.com/user-attachments/assets/18dadb7b-16b7-4cc9-8fac-323e083e0247)

```c++
  const QString itemStyle = R"(
      QListWidget {
          background: white;
          color: black;
          outline: none;
          padding: 0px;
          margin: 0px;
      }

      QListWidget::item {
          position: absolute;
          padding: 2px;
          margin: 0px;
          border: none;
          border-radius: 5px;
          outline: none;
          background-color: transparent;
      }

      QListWidget::item:hover {
          background-color: #c0e0f0;
      }

      QListWidget::item:selected {
          position: absolute;
          background-color: #c0e0f0;
          border: 1px solid #4169e1;
          color: black;
          margin: 0px;
          padding: 0px;
      }

      QListWidget::item:selected:hover {
          border: 1px solid black; /* Black border when selected and hovered */
      }

      QListWidget::item:focus {
          outline: none; /* Remove the default focus outline */
      }
  )";

  CompComps->setStyleSheet(itemStyle);
```

ra3xdh, 2024-07-30:
@dsm, everything looks good on Linux. You may open PR with the corrected style sheet.

wawuwo, 2024-07-30:
I have the same behaviour (no "highlighting" when mouse hovers over a component icon) on MacOS build with QT 6.7.0 and on Debian with QT 6.4.2

ra3xdh, 2024-07-30:
@wawuwo , Please test the proposed patch with fixed CSS for `CompComps` widget.

dsm, 2024-07-30:
I'll sent a PR.

wawuwo, 2024-07-30:
Highlighting actually looks very well with the proposed fix, both on Debian + QT 6.4.2 and MacOS + QT 6.7.0, though some components have a strange white border.

[highlight-on-hover-debian-qt-6.4.0.webm](https://github.com/user-attachments/assets/10c65749-76e0-42c0-9afd-9c084d94aed5)

https://github.com/user-attachments/assets/6144c3e1-848c-4807-9ad5-d12181f12b17

ra3xdh, 2024-07-30:
> though some components have a strange white border.

These devices have manually drawn SVG icon. Other devices have auto-generated icon. I think the issue with white border may be ignored.

dsm, 2024-07-30:
Yes I also notice that white border.

dsm, 2024-07-30:
I sent PR I changed something, Is there any problem.

ra3xdh, 2024-07-30:
Fixed by #877

PR #877 (2024-07-30): fixed component highlighting.
Hi, this PR fix #874.
closed 2024-07-30 by ra3xdh