ra3xdh/qucs_s #875: Select-all does not select wires on the boundary

hvwyl, 2024-07-28:
Select-all operation won't select wires on the boundary

[image]

wawuwo, 2024-07-30:
@hvwyl, hi! What version of Qucs-S do you use? Do you remember by any chance the last version without this bug? I am able to reproduce the bug in head-of-the-branch build, just need some more info to investigate the cause

hvwyl, 2024-07-30:
> @hvwyl, hi! What version of Qucs-S do you use? Do you remember by any chance the last version without this bug? I am able to reproduce the bug in head-of-the-branch build, just need some more info to investigate the cause

I have fixed it by changing the properties of `QMargins` in `qucs_actions.cpp`. Although this may not the best solution.

[image]

wawuwo, 2024-07-31:
@hvwyl I've found the exact place where this bug was introduced, it's the 8c584fc4. In that commit I used `QRect.bottom()` and `QRect.right()` to pass bounds of selection rectangle, but these methods return a value which is smaller than the actual one by one unit ([here](https://doc.qt.io/qt-6/qrect.html#bottom) and [here](https://doc.qt.io/qt-6/qrect.html#right)).

I've opened PR #878, it doesn't *directly* address this bug, but should fix the problem (at least it did while I was testing it).

ra3xdh, 2024-08-03:
Fixed by #878

closed 2024-08-03 by ra3xdh