ra3xdh/qucs_s #883: Path to Octave application not retrieved from windows registry in version 24.3.0

peterbaltus, 2024-08-08:
Version 24.3.0 windows, installed using .exe installer over previous version
Symptom: opening Octave window from view menu does not start Octave

Analysis: seems to be caused by the "Octave Path" in the Application Settings of the file menu (tab "Locations") being empty. Inserting the correct path starts Octave. However, quitting and restarting Qucs-s results in the "Octave Path" being empty again. Apparently the setting is not saved across restarts of Qucs-s

After some poking around I found that the registry key "The key in the registry is Computer\HKEY_CURRENT_USER\SOFTWARE\qucs\qucs_s\OctaveExecutable" does hold the correct path, it seems that this is not retrieved when starting Qucs-s (?)

Work-around: reverted to Qucs-s 24.2, which works correctly.

This is my first issue post on Github, apologies if it is not in the correct format/not complete etc. If so, please let me know and I will try to update it.

    Peter

ra3xdh, 2024-08-12:
Yes, I confirm the same issue for Linux. The octave starts after setting the path to `/usr/bin/octave-cli`, but this path didn't saved between the application restart. This has appeared after migrating the settings to new API. I will provide a some fix soon.

The existing Octave window was designed for Octave version 3. This Octave version had no GUI. In fact the Octave Dock is a reduced functionality terminal emulator. This dock window doesn't provide a full access to all features of modern Octave. I am considering to remove this home-made terminal emulator in the future versions and start Octave in GUI mode from Tools menu instead. I am also considering to totally remove the possibility for starting Octave from within Qucs-S. Everything could be done using system main menu and Octave could be started using modern GUI mode. And I would ask for feedback why is this dock still in use?

ra3xdh, 2024-08-12:
See also #887 for proposed replacement for Octave dock.