ra3xdh/qucs_s #892: Strange issues as well as "Ghost" data points when plotting

tomhajjar, 2024-08-16:
The past week I noticed a number of old projects don't work the same anymore and there are "ghost" data points being plotted in areas they shouldn't be in. Attached is an old project as saved in a backup. Other projects had the same issue.

By chance I used AC and S-parameter analysis for the same schematic.  

1. Ghost data points are present as shown in attached screen shots

2. If I run an analysis, the data is garbage and nothing I do will make the new data look like the old. If I grab my back-up and over write the entire project the analysis suddenly works. I cannot explain what is happening. As I said other projects had the same issue.

[image]
[image]

[Transmission_Line_prj.zip](https://github.com/user-attachments/files/16642473/Transmission_Line_prj.zip)

ra3xdh, 2024-08-19:
I confirm this problem. The virtual point at (0,0) is always rendered. All *point-type* plots are affected. I will provide a fix soon. The most probably I will prepare v24.3.1 release in September containing this fix, because the reported bug is critical.

ra3xdh, 2024-08-20:
Fixed by #897

PR #897 (2024-08-20): Point type graph redering fix
This PR fixes #892. Introduced `Graph::ScrPt::type` to hold the information about point type
closed 2024-08-20 by ra3xdh