ra3xdh/qucs_s #896: ngspice compatible Library of Transmission Line components

ra3xdh, 2024-08-20:
### Discussed in https://github.com/ra3xdh/qucs_s/discussions/895

<div type='discussions-op-text'>

<sup>Originally posted by **tomhajjar** August 19, 2024</sup>
I've been working on some Transmission Line components that will work with ngspice. Most of the components use the Spice TLIne. I also found a lossless directional coupler that I made into a subcircuit, 

Attached is the project and the Library

[image]
[image]
[image]

[Transmission_Line_Lib_prj.zip](https://github.com/user-attachments/files/16662657/Transmission_Line_Lib_prj.zip)
[Transmission_Line_prj.zip](https://github.com/user-attachments/files/16662655/Transmission_Line_prj.zip)</div>

ra3xdh, 2024-10-24:
The library uploaded by #1019

closed 2024-10-24 by ra3xdh

tomhajjar, 2024-10-25:
@ra3xdh 

I updated Transmission_Line.lib on 10/25/24

Added 180 deg Hybrid Coupler and added better descriptions of the devices.

[image]

[Transmission_Line_prj.zip](https://github.com/user-attachments/files/17526801/Transmission_Line_prj.zip)