ra3xdh/qucs_s #900: ngspice compatible Library of Digital components for mixed signal simulations

tomhajjar, 2024-08-22:
Is there any thought about adding an ngspice compatible digital library? ngspice is compatible with PSpice/MicroCap digital libraries.

I spent a lot of time testing mixed signal simulations mixing analog and digital components. Attached are two projects. First tests digital models and the second is a fairly complex mixed analog-digital simulation.

[image]
[image]

[Digital_Analog_Mixed_prj.zip](https://github.com/user-attachments/files/16702687/Digital_Analog_Mixed_prj.zip)
[Trainline_Driver_prj.zip](https://github.com/user-attachments/files/16702688/Trainline_Driver_prj.zip)

tomhajjar, 2024-08-22:
The Digital Library(s) could be differentiated by the source of the Spice models. 

The first Library would be derived from "MicroCap_dig000.lib". Examples of using this library to make models would be given so we wouldn't need to make every 74XX, 74HCXX, 74SXX...etc device. I have made a number of devices. All require "set ngbehavior=psa"

The guys at ngspice converted 5 devices from the LTSpice digital libraries into XSPICE devices and named it "74HCng_gates.lib". It would require a massive amount of work to convert all the devices in the the LTSpice digital libraries.

The late Jim Thompson created "Digital_Behavioral.lib". We could use these 11 devices (normalized vcc=1) in another library. Jim also created a number of other models.

Unfortunately TI made very few "Transient/Behavioral" digital models so the Library would be tiny.

tomhajjar, 2024-08-23:
First cut at a Digital Library based on "MicroCap_dig000.lib". I added Requirements to the Show model window.
I need suggestions of additional devices to add to the Library. 

[image]
[image]

[Digital_Library_1_prj.zip](https://github.com/user-attachments/files/16720318/Digital_Library_1_prj.zip)

ra3xdh, 2024-08-23:
Thanks! I will evaluate the libraries and post feedback in the next few days. At least MicrCAP 74HC library may be included in the next release. It's need at least to add a few ICs the most commonly used in analog circuits. 

The best 74HC library for analog simulation mode would be this one: https://github.com/texane/power_inverter/blob/master/ltspice/logic/74hc/74hc.lib This library contains not a simple delay-type mode, but a full transistor level I/O buffer schematic model. Unfortunately it contains incompatible LTspice extension for represent the logic gate. This library requires significant effort to be ported to Ngspice. I don't consider this to be done by current development team. This task is not complex, but time consuming.

tomhajjar, 2024-08-23:
74HCng_short_2.lib in the ngspice examples is a conversion of 5 devices in 74hc.lib. The conversion doesn't include a full transistor level I/O buffer schematic model.

tomhajjar, 2024-08-27:
The following MicroCap Digital Libraries are included in "Digital_Analog_Mixed_prj.zip". We need to choose what devices should be made into Qucs-S Libraries. Presently "Digital_Library_1.lib" only has eight 74HC devices.

MicroCap_dig000.lib
MicroCap_74lv.lib
MicroCap_cmos.lib
MicroCap_ecl.lib

Back in the day ECL and CMOS gates devices were used as Analog components.

tomhajjar, 2024-09-26:
@ra3xdh 
Made a PBRS source using MicroCap_74lv.lib. I would have used MicroCap_dig000.lib but it didn't have a 74HC164. Didn't want to convert a 74LV164 to 74HC164 so just used the LV library. 

FYI:
No device's outputs work unless they have a "load". 
No being able to specify a device in a large library requires copying the device and pasting it into a new file. 

[image]

[Digital_PRBS_Generator_prj.zip](https://github.com/user-attachments/files/17157113/Digital_PRBS_Generator_prj.zip)

ra3xdh, 2024-10-29:
I have assigned one of my students for porting digital libraries from MC/LTSpice to Qucs-S+Ngspice. Let's see what happens.

tomhajjar, 2024-10-29:
Post some of the early work so I can check it. No reason to go too far if mistakes can be found early

tomhajjar, 2024-10-30:
@ra3xdh 

One issue with these new digital libraries is they don't work with existing digital source, digital simulation, timing diagram, truth table, logic 1/0..etc.

Any thoughts on making a new digital source using the Spice PWL source like the attached project or use "rectangular voltage" to make a clock?

Also making a new timing diagram using VCC as the "scaling/spacing" factor like done in ngspice digital examples as attached?

[image]

[Digital_Test_prj.zip](https://github.com/user-attachments/files/17564825/Digital_Test_prj.zip)

ra3xdh, 2024-10-30:
> One issue with these new digital libraries is they don't work with existing digital source

Digital source works with analog mode. See the attached screenshot.

The digital simulation should not work with analog mode. It is performed only using VHDL/Verilog. The timing diagram and truth table also accepts data only from Verilog/VHDL backend. It is hardly possible to convert SPICE dataset to acceptable data. 
[image]

tomhajjar, 2024-10-30:
Why am I only getting one cycle of CLK? Do I have to add additional data points?

The "fake" timing plot doesn't look bad...

[image]

[Digital_PRBS_Generator_prj.zip](https://github.com/user-attachments/files/17572239/Digital_PRBS_Generator_prj.zip)

tomhajjar, 2024-10-30:
I updated Digital_Analog_Mixed_prj. A folder has the ngspice netlists used by DuSpice. The plots use the method used in ngspice digital examples.

[image]
[image]

[Digital_Analog_Mixed_prj.zip](https://github.com/user-attachments/files/17577480/Digital_Analog_Mixed_prj.zip)

ra3xdh, 2024-12-02:
The implementing of this feature continued in #1034. Closing, because two issues are duplicating each other.

closed 2024-12-02 by ra3xdh