ra3xdh/qucs_s #902: Qucs_S Windows crashes when renaming missing sub-circuits

tomhajjar, 2024-08-23:
I renamed some sub-circuits and when I try to update the original schematic with the new sub-circuit names, Qucs-S Windows crashes.

Deleting the "old" sub-circuit and replacing the "new" is not desirable since all the original sub-circuit parameters are lost.

I don't remember entering a new name for a sub-circuit being an issue before.

[image]
[image]
[image]

[BFP193W_amplifier_prj.zip](https://github.com/user-attachments/files/16732412/BFP193W_amplifier_prj.zip)

ra3xdh, 2024-08-23:
I confirm this bug. I will provide a fix soon.

tomhajjar, 2024-08-24:
Any idea why the gain of your original circuit is soo low? It should >20dB not <7dB.

[image]

ra3xdh, 2024-08-24:
> Any idea why the gain of your original circuit is soo low? It should >20dB not <7dB.

I cannot remember where I took this schematic. Maybe the microstrips has a wrong design.

ra3xdh, 2024-08-27:
Fixed by #918

closed 2024-08-27 by ra3xdh