ra3xdh/qucs_s #905: Matching Circuit Tool generates wrong results

tomhajjar, 2024-08-23:
In testing Matching Circuit Tool for an "L-section match" case, I found the component values to be unrealistic. 

The input match L1 shunt inductor value of "0.000H" should be interpreted as a short to ground or using "creative license", an open circuit to ground. The output match L4 series inductor value of "2.556H" can only be interpreted as an open circuit.

I didn't test any other cases since having to copy and paste S-parameter values becomes tedious and error prone.

The "new"

[image]
[image]
[image]

[BFP193W_amplifier_prj.zip](https://github.com/user-attachments/files/16734237/BFP193W_amplifier_prj.zip)

andresmmera, 2024-08-24:
Hi @tomhajjar , thanks for pointing this out.

The device is not unconditionally stable and it cannot be conjugately matched. Equations 12.40a/b in [1] don't have a solution for this case since the argument of the square roots is negative.

[image]

[image]

I think that the best solution would be to check if the device is unconditionally stable [here](https://github.com/ra3xdh/qucs_s/blob/current/qucs/dialogs/matchdialog.cpp#L976) in order to avoid wrong results like this. I can write a fix for this later.

Schematic: 
[BFP193W_spar_analysis.zip](https://github.com/user-attachments/files/16735773/BFP193W_spar_analysis.zip)

[1] Microwave Engineering. David M Pozar. 4th Edition. 2012

tomhajjar, 2024-08-24:
Not often a device's Spice model correlates so well with it's S-parameter data...

[image]
[image]

tomhajjar, 2024-08-24:
@andresmmera

One minor issue with the tool is if any entry or setting is "wrong" or out of bounds, a error message is shown and the tool is closed. Is there a way to allow the user to fix the wrong entry and not close the tool?

andresmmera, 2024-08-24:
Good point. I'll take a look later.

ra3xdh, 2024-08-24:
Fixed by #907. Closing as completed.

closed 2024-08-24 by ra3xdh

andresmmera, 2024-08-24:
> @andresmmera
> 
> One minor issue with the tool is if any entry or setting is "wrong" or out of bounds, a error message is shown and the tool is closed. Is there a way to allow the user to fix the wrong entry and not close the tool?

Fixed in PR #909

tomhajjar, 2024-08-25:
This Qorvo tool is useful for seeing the effect of R on K.

[image]