ra3xdh/qucs_s #908: Simulation fails after adding a matching network

tomhajjar, 2024-08-24:
After adding a matching network a BFP193W amplifier simulation fails. The device has no collector bias voltage and I cannot see what is causing the issue.

[image]
[image]
[image]

[BFP193W_amplifier_prj.zip](https://github.com/user-attachments/files/16736937/BFP193W_amplifier_prj.zip)

ra3xdh, 2024-08-24:
It is not Qucs-S bug. The R1 resistor has no DC path to ground and Ngspice DC OP simulation fails. The schematic works as expected after adding 1Meg resistor between R1 and ground. Unfortunately Ngspice error message doesn't help to debug the schematic. 
[image]

closed 2024-08-24 by ra3xdh

tomhajjar, 2024-08-24:
Vc being wrong threw me off...