ra3xdh/qucs_s #921: qucs-s for OSX 10.14

bnilsson11, 2024-09-02:
In qucs-s 24.3.0 for MacOS 10.14, the Delete command using keyboard Delete or Backspace is not working.
Delete using mouse right-click is working.

dsm, 2024-09-02:
I'll fix this issue, normally delete shortcut is QKeySequence::Delete but delete button not working for macbook m3 pro so I changed QKeySequence::Delete to QKeySequence::Backspace but in qt 6.2 Meta+H is backspace but Qt 6.7 Meta+H or backspace is KeySequence::Backspace. So in your mac Mata+H should work it qt says. but I'll added QKeySequence::Backspace and QKeySequence::Delete both for delete. can you test it ?

bnilsson11, 2024-09-02:
This Mac is at my old work place, and I am there every Monday morning to after lunch time.
So it will take a week before I can test it.

dsm, 2024-09-02:
Ok I'll add directly backspace and delete key and I don't use QKeySequence and I'll test it both macos and logi mx keys keybords as soon as I get home.

dsm, 2024-09-02:
I pushed a commit that solve this issue, I tested backspace and delete keys both works under mac, some mac keyboards don't have delete key so backspace added for this reason.

ra3xdh, 2024-09-04:
Fixed by https://github.com/ra3xdh/qucs_s/commit/13e920915358bb7b56ff30f1fc6917b6921b01ef Closing as completed.

closed 2024-09-04 by ra3xdh

bnilsson11, 2024-09-09:
Where can I find the release for MacOS 10.14 with working Delete/Backspace?

dsm, 2024-09-09:
You can download latest 24.3.1 macosx release in release section

bnilsson11, 2024-09-09:
Thanks, now working fine. 
Good work.