ra3xdh/qucs_s #922: Relative user library path requested

bnilsson11, 2024-09-02:
When working on the same project from different sites, it would be convenient to use the same user_lib/libraries.
However, the library device symbol is using full path, and this is usually not same for different workplace sites.
So a circuit with a user lib device defined at one workplace is broken as seen from another workplace.
Allowing paths relative to qucs_s home would solve this.

bnilsson11, 2024-09-02:
I cannot get it to work.
Where should I place my project libs?

ra3xdh, 2024-09-02:
Place the library inside the project directory. You should create the project first. The projects are located by default in `$HOME/QucsWorkspace` (new installations) or `$HOME/.qucs` (old installations).

ra3xdh, 2024-09-02:
I have just checked this feature (#567). Everything works on my machine.

bnilsson11, 2024-09-02:
Ok, it is working, I forgot to open the project first.

Back to relative paths, just interested:
Search of user_lib relative to qucs home, was it ever implemented?

ra3xdh, 2024-09-02:
The libraries from `$HOME/QucsWorspace/user_lib` always share absolute path, because the system and user libraries may have the same name. The introduction of the project libraries with relative path is sufficient for sharing libraries with projects and schematics.

ra3xdh, 2024-09-02:
The `$HOME/QucsWorkspace` is the default project directory path since v24.2.1. Closing this issue as completed.

closed 2024-09-02 by ra3xdh