ra3xdh/qucs_s #924: ST Triac library

dsm, 2024-09-03:
Hi, will this library add for qucs_s with ngspice 
 https://www.st.com/resource/en/spice_model/standard_snubberless_triacs_pspice.zip

dsm, 2024-09-03:
Can MOC3052 and MOC3063 be added from https://fotoelektronika.com/spice-models/ ?

ra3xdh, 2024-09-03:
> Can MOC3052 and MOC3063 be added from https://fotoelektronika.com/spice-models/ ?

Yes, the Optocoupler.lib could be extended. I have added this task to #846

dsm, 2024-12-11:
```spice
.subckt BTA41-800B A K G
X1 A K G Triac_ST params:
+ Vdrm=800v     
+ Igt=100ma
+ Ih=80ma        
+ Rt=0.01
+ Standard=1
* 1999 / ST / Rev 0
.ends
```

how I can use this in qucs_s psice library

ra3xdh, 2024-12-11:
The best way is to use *Spice Library device* See the chapter 8.2 of the latest Qucs-S tutorial: https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf  There exists symbol patterns for thyristor devices. 

The works on extension of this feature have been planned. See #944

ra3xdh, 2025-01-08:
I have added generic ST Triac model in `Thyristor.lib` by #1197. New device is named *TriacGeneric*. But I failed to port this device to pure Ngspice syntax. This device requires setting the LTSpice compatibility mode in *Simulation->Simulator settings*

closed 2025-01-08 by ra3xdh

ra3xdh, 2025-01-19:
See also #1210 This adds a way to create a library from parametric subcircuits like ST triacs. A sample library containing three ST devices is attached.

dsm, 2025-01-19:
Thanks I tested it works as expected :)