ra3xdh/qucs_s #929: qucs-s windows uses old icon mac uses new one

dsm, 2024-09-04:
Hi, this is the difference mac and windows icon and qucs-s.exe normally uses different icon but innosetup changed icon to old icon but not visible in explore it's strange. We should set app icon using Qt API and CMake. I created new.ico based on qucs-s.svg in qucs/bitmaps .

[image]

ra3xdh, 2024-09-04:
It's need to replace icons in `contrib/InnoSetup/misc` It still use 64x64 icons. Maybe switch to higher resolution or vector icon if supported.

dsm, 2024-09-04:
256x256 looks good on Full HD screen i changed in innosetup locally and build exe, Qt page says convert icon to ico file, i think svg not supported.

dsm, 2024-09-04:
ico file can store multiple resoultion icons in a single file using

magick convert 16.png 32.png 48.png 128.png 256.png -colors 256 icon.ico

dsm, 2024-09-05:
I sent a PR #932 to fixed windows icon but app title icon not changed it needs to be changed in Qt code
App icon:
[image]

title icon:
[image]

closed 2024-09-05 by ra3xdh

dsm, 2024-09-05:
#ifndef __APPLE__
  setWindowIcon (QPixmap(QString(":/bitmaps/hicolor/scalable/apps/qucs.svg")));
#else

I tested this usage and succesfully render svg. about diolog also used 256x256 png maybe this one can be used there.

dsm, 2024-09-05:
```cpp
#ifndef __APPLE__
  setWindowIcon (QPixmap(QString(":/bitmaps/hicolor/scalable/apps/qucs.svg")));
#else
```

I tested this usage and succesfully render both app and title svg icon. about diolog also used 128x128 png maybe this one can be used there. Why you think the existing icon should remain (big.qucs.xpm) generally app and title icons are same.

ra3xdh, 2024-09-05:
> I tested this usage and succesfully render both app and title svg icon. about diolog also used 128x128 png maybe this one can be used there.

Yes, this one could be accepted.

PR #932 (2024-09-05): update app icon to new one.
Hi, this PR partially fixed #929

- update app icon to new one.
- added icon to exe setup.