ra3xdh/qucs_s #930: Hotkey conflict of Ctrl+W on Linux (AppImage)

SmallJoker, 2024-09-04:
The hotkeys of File -> Close and Insert -> Wire both use Ctrl + W. Neither hotkey is triggered when pressing Ctrl + W.

**Reproduced on:**

* Ubuntu 22.04 with xfce4
* Debian Trixie (13) with openbox

**Known good version:** 2.1.0 (Oct 2023), which uses Ctrl+E to insert a wire
**Known bad version:** 24.2.1 (March 2024)

**Not affected:**

* Windows

Suggestion: use Ctrl + E for wire placement or unbind the File -> Close hotkey (but Ctrl + W is commonly used for this function).

dsm, 2024-09-04:
fileClose->setShortcut(QKeySequence::Close); problem is this line I think we should use exact keys, QKeySequence usage changed platform to platform .

ra3xdh, 2024-09-04:
Yes, it would be better to hardcore shortcut to avoid conflicts.

ra3xdh, 2024-09-05:
Fixed by #931

closed 2024-09-05 by ra3xdh

SmallJoker, 2024-09-05:
Wow that was quick. Thank you, @ra3xdh.

ra3xdh, 2024-09-05:
I would also remember that AppImage is not a recommended way to get the package for Debian/Ubuntu. This packaging method may have limitations. AppImages should be used only if you have no administrative rights on the Debian machine or need the latest continuous release build. The DEB/RPM packages are preferable. I am distributing the DEB/RPM packages for the latest stable release.