ra3xdh/qucs_s #941: Update problem due to filesize error - Qucs-s 24-3-1 - Debian Unstable

btorrem, 2024-09-10:
Hello,
I get the following error during an upgrade :

Get:1 http://ftp.gwdg.de/pub/opensuse/repositories/home%3A/ra3xdh/Debian_Unstable  qucs-s 24.3.1-1 [4,373 kB]
Err:1 http://ftp.gwdg.de/pub/opensuse/repositories/home%3A/ra3xdh/Debian_Unstable  qucs-s 24.3.1-1
  File has unexpected size (4375060 != 4373196). Mirror sync in progress? [IP: 2a06:93c7:f0::f79 80]
  Hashes of expected file:
   - SHA256:21ff895ea7d4a9b230011916db74d2ba3391ed83d624ff0ea66680259a912006
   - SHA1:1c17af6ccd63747912f56a80e4e1057ef2afc125 [weak]
   - MD5Sum:3053a8890155680b7563690cac31e60c [weak]
   - Filesize:4373196 [weak]
Error: Failed to fetch http://ftp.gwdg.de/pub/opensuse/repositories/home%3A/ra3xdh/Debian_Unstable/amd64/qucs-s_24.3.1-1_amd64.deb  File has unexpected size (4375060 != 4373196). Mirror sync in progress? [IP: 2a06:93c7:f0::f79 80]

ra3xdh, 2024-09-10:
This very strange error appears with OBS repositories time to time. I have recently checked Ubuntu-22.04 repository and it works as expected. I cannot figure out what causes this error. I have to to install Debian-12 in the clean VM to reproduce this error. You may try clearing caches and/or disabling and enabling repository again.

I would recommend to disable the repository, download DEB, and install it with `dpkg`. Use the *Grab binary package directly* link at this page: https://software.opensuse.org/download.html?project=home%3Ara3xdh&package=qucs-s

btorrem, 2024-09-11:
Hi, the upgrade did work today. 
I only did an apt-get autoclean before doing the update followed by an upgrade. 
Strange behaviour indeed, but it's OK now.

ra3xdh, 2024-09-11:
I have just checked Debian-12 repository using fresh VM. Installation works as expected. This issue may be closed as self-fixed.

closed 2024-09-11 by ra3xdh