ra3xdh/qucs_s #942: Laser Diode Simulations

tomhajjar, 2024-09-11:
Dug out a Laser Diode simulation I did in the 1990s on a Sony device. I also downloaded some LTspice simulations. Should be useful for someone designing low power laser circuits. Rohm has Spice models for some of it's diodes. 

The Sony SLD1121VS model was done by Bob Pease in EDN. 
Alexander Bordodynov made a number of "generic" devices.
Also included is an Intusoft model.

[image]
[image]

[Laser_Diode_prj.zip](https://github.com/user-attachments/files/16956276/Laser_Diode_prj.zip)

ra3xdh, 2024-09-11:
Thanks! I will make a Laser Diode library from these models and add it in the next release.

ra3xdh, 2025-01-05:
Laser diodes library uploaded by #1185

closed 2025-01-05 by ra3xdh