ra3xdh/qucs_s #944: Lib device limitation

tomhajjar, 2024-09-12:
Why can't the "new" Lib device be used to create a Library? Whey I try to make a Library from sub-circuits made using the Lib device, The Library is made missing all the Spice model information. I don't remember this limitation.

[image]
[image]
[image]
[image]

ra3xdh, 2024-09-12:
The adding of subcircuit extraction while creating libraries from SpiceLibComp devices is not a straightforward task. This device is only referencing library in the netlist. Don't expect this to be added in the near future. Use old *Spice File* device  for creating libraries. 

On the other hand the generic laser diode device need to be converted into the unified subcircuit. The device contains only standard diode devices and controlled sources. This will make this device available for both Qucsator and SPICE.

tomhajjar, 2024-09-12:
Will the Spice netlist device be "updated" to have the same "new" features as the Spice library device? 
The new window has features like choosing Symbol files and a subcircuit viewer that would be useful in making large Libraries.

[image]

ra3xdh, 2024-09-12:
I don't consider updating of the SpiceFile device. It has different design and works differently.

ra3xdh, 2024-09-13:
I can implement two things to improve the new *Spice Library Device*

* Add option like *Embed subcircuit in netlist* This option must be active when creating library. Otherwise the netlist will be referencing library via `.INCLUDE` directive. It is not clear how to extract complex subciruits containing subcircuits and `.MODELS` itself from the library file. The internal subcircuit definitions are placed before the closing `.ENDS` directive.
* Maybe add possibility to paste subcircuit code in *Spice Library Device* properties window and combine with the previous proposed option. Make the *Subcircuit* editor editable. This will resolve the problem with extraction of the subcircuit from library.  
* The *Spice File Device* can be used with the user libraries placed in `$HOME/QucsWorkspace/user_lib`. The symbols are placed in a separate subdirectory and could be loaded automatically. Look how the `XyceDigitalDevices` library is made.  This method is not implemented from GUI. It's need to provide a way to import SPICE libraries in a such way using GUI tools.

The old *Spice File* device could be kept only for backward compatibility after the proposed improvements will be done. It will be needed only for legacy Qucsator simulations.

tomhajjar, 2024-09-13:
Where do the symbols for XyceDigitalDevices come from?

ra3xdh, 2024-09-14:
> The Digital examples in the XYCE example folder all have the wrong path for XyceDigitalDevices.lib. 

To be honestly I have never tested these examples. This was made by Mike Brinson for Xyce v6 and may not work anymore with the latest Xyce version. 

> This is a different type of Library that I didn't know Qucs-S supported. 

Qucs-S supports such libraries. The *Spice Library Device* has undocumented feature of attaching symbol to SPICE library. The SPICE SUBCKT and symbol file should have the same name. Library manager looks in a special directory and loads an appropriate symbol. This feature is not available from GUI. Also read more here: https://qucs-s-help.readthedocs.io/en/latest/SubLib.html#usage-of-the-whole-spice-library 

Something may become broken after the recent redesign of SpiceLibComp, but this feature requires redesign itself. This is another development task. I don't recommend to use this feature for libraries design until I will provide an improved version. Again don't expect a quick solution here.

>  How are the symbols for XyceDigitalDevices.lib stored in the path below pointed to by the library?

The XyceDigital library was made manually with the text editor by Mike Brinson around 2016. The symbols were probably copied from some schematic. I don't posses the source schematic for the symbols.

tomhajjar, 2024-09-14:
The XYCE digital examples all work when you correct the Library path. I used a text editor and search-replace to fix them. 

The PATH however assumes you used a 32bit version of Windows and used the Qucs-S Installer that placed the files in the "C:/Program Files (x86)/Qucs-S/" folder.

The symbols can be opened in Qucs if you add the line "<Qucs Schematic 24.2.1>"

[image]
[image]
[image]
[image]

tomhajjar, 2024-09-14:
I converted all the XyceDigitalDevices.lib symbols to "sch" by adding <Qucs Schematic 1.0.0> to the first line. 

There is no consistency to symbol colors, body size, PIN position or having even having Ports. Can you explain why AND2 has no ports?
[Digital_xyce_prj.zip](https://github.com/user-attachments/files/17002836/Digital_xyce_prj.zip)

ra3xdh, 2025-01-18:
I am working on solution. This will be implemented in the upcoming v25.1.0 release.

closed 2025-01-19 by ra3xdh

tomhajjar, 2025-01-19:
I will make a project to test the new functionality.