ra3xdh/qucs_s #946: XyceDigital.lib symbols and test files

tomhajjar, 2024-09-14:
I cleaned up the XyceDigital.lib symbols and test files. 

We still have the PATH issue because the library devices in a schematic use an explicit PATH pointing to the folder where XyceDigital.lib is located. Also don't know how Qucs grabs the symbols for XyceDigital.lib, I assume it's hard coded.

[Digital_xyce_prj.zip](https://github.com/user-attachments/files/17003683/Digital_xyce_prj.zip)
[XyceDigital_symbols.zip](https://github.com/user-attachments/files/17003684/XyceDigital_symbols.zip)

ra3xdh, 2024-09-15:
> I cleaned up the XyceDigital.lib symbols and test files.

Thanks! I will upload updated examples. 

> Also don't know how Qucs grabs the symbols for XyceDigital.lib, I assume it's hard coded.

It is not hardcoded. Qucs looks for symbols in the directory named the same as library. For example in `XyceDigital` for `XyceDigital.lib`. If the symbol `XyceDigital/AND2.sym` is found for `.SUBCKT AND2 ...` it will be loaded automatically. Otherwise the default square symbol will be created. There is no way to specify the pins sequence using this method. Please don't use this method for making new libraries. It requires a redesign. We have implemented only one library using this feature.

ra3xdh, 2024-09-15:
> We still have the PATH issue because the library devices in a schematic use an explicit PATH pointing to the folder where XyceDigital.lib is located

This is fixed in v24.3.1 by #931

ra3xdh, 2024-09-15:
The updated libraries uploaded by #947

closed 2024-09-15 by ra3xdh

tomhajjar, 2024-09-20:
@ra3xdh 

The latest nightly has the old XyceDigital symbols

ra3xdh, 2024-09-20:
This build is compiled from another branch and contains only fix for the bugs reported in #951 The upcoming v24.3.2 will contain no library update, only this two bugfixes. The library update will be included in 24.4.0 (planned for November or December)