ra3xdh/qucs_s #948: .INCLUDE statement Filename Edit/Browse doesn't work - altermod usage

tomhajjar, 2024-09-15:
Having two issues:

1) .INCLUDE statement manually editing Filename or using Browse doesn't work anymore. I tried manually editing the Spice filename and file info goes blank. I then tried using Browse and again file info goes blank. You can do all the steps, click "Apply" and everything looks good but when you click "OK" all fields go blank. In fact make no changes and click "OK" and all all fields go blank.

For simplicity, I decided to shorten the filename "hicumL2V2p40_model_card.lib" -> "hicumL2V2p40.lib" and found the issue.

2) I wanted to reproduce the data in the ngspice netlist "hic2_out.sp" and can't figure out how to get the line "altermod @hicumL2V2p40[flsh]=1" to work in Qucs. 

[image]
[image]
[image]
[image]
[image]
[image]

[HICUM2_prj.zip](https://github.com/user-attachments/files/17007711/HICUM2_prj.zip)

ra3xdh, 2024-09-16:
> INCLUDE statement manually editing Filename or using Browse doesn't work anymore. 

I confirm this bug. There is no workaround. This bug is critical. I will prepare v24.3.2 release containing the fix as soon as possible.

tomhajjar, 2024-09-16:
I will use the text editor work-around.

Can you show how to get "altermod @hicumL2V2p40[flsh]=1" in the ngspice netlist "hic2_ib_ic_thermal.sp" to work in Qucs?

ra3xdh, 2024-09-17:
> Can you show how to get "altermod @hicumL2V2p40[flsh]=1" in the ngspice netlist "hic2_ib_ic_thermal.sp" to work in Qucs?

There is no way except composing a custom Nutmeg script. I have deprecated the explicit `SweepModel` property of the Parameter sweep. See #839 But I didn't take into account that the model may be defined outside schematic in the library file using `.include`. There is not way to substitute the variable for model defined in the external file.

tomhajjar, 2024-09-19:
I took the easy way out and made a 2nd Spice model file with "thermal" enabled.

[image]

[HICUM2_prj.zip](https://github.com/user-attachments/files/17051938/HICUM2_prj.zip)

ra3xdh, 2024-09-19:
The both issues are fixed on this branch https://github.com/ra3xdh/qucs_s/tree/release/24.3.2  I will try to prepare the 24.3.2 update in the next few days. I have added model sweep pattern recognition. 

[image]

tomhajjar, 2024-09-19:
Will it work for both cases like xyce?

[image]
[image]

[BJT_Beta_sweep_xyce.zip](https://github.com/user-attachments/files/17062987/BJT_Beta_sweep_xyce.zip)

ra3xdh, 2024-09-23:
Fixed by #955

closed 2024-09-23 by ra3xdh

tomhajjar, 2024-09-23:
Does the latest nightly fix the Beta sweep of a Red BJT?