ra3xdh/qucs_s #951: Spice netlist versus .INCLUDE Issues

tomhajjar, 2024-09-18:
Discussion #950 has found a few more issues. Attached are two schematics showing the issues. 

1) The SB5100.lib diode Spice model doesn't work unless it is made into a sub-circuit using "Spice netlist". See SB5100_recovery.sch.

2) When making the subcircuit SB5100_sub.sch, the 1ST time you make it the filename has the entire PATH. When you place this sub in the schematic, the simulation fails and you find the "Spice netlist" is missing. See . You have to re-open SB5100_sub.sch and Browse again for SB5100.lib. I made SB5100_sub_1st.sch to show the problem.

[image]
[image]
[image]

[Rectifiers_prj.zip](https://github.com/user-attachments/files/17038353/Rectifiers_prj.zip)

ra3xdh, 2024-09-18:
The SB5100 diode model works on my machine. See https://github.com/ra3xdh/qucs_s/discussions/950#discussioncomment-10682790 

> the 1ST time you make it the filename has the entire PATH. When you place this sub in the schematic, the simulation fails and you find the "Spice netlist" is missing. 

The application applies relative path conversion after #640 The path is set relatively to project directory or containing schematic directory. This may be fixed as the following:

* If the schematic doesn't belong to the project set absolute path
* If schematic belongs to project set relative path

tomhajjar, 2024-09-25:
Still having the issue in item #2 above. I have to redo the sub-circuit twice to get rid of the extended path. I didn't document it this time.

ra3xdh, 2024-09-25:
I will implement the path setting policy proposed here: https://github.com/ra3xdh/qucs_s/issues/951#issuecomment-2358983233 in the next release.

ra3xdh, 2024-10-24:
Fixed by #1018

PR #1018 (2024-10-24): Apply path conversion only for saved schematic or schematic in projects
This PR fixes #951. The path conversion is applied only for saved schematics or schematics that belong to some projects. If schematic is not saved only full path may be substituted, because the schematic could be moved elsewhere. 

The better solution would be to automatically convert relative paths when schematic is saved. May be implemented only after #974 because there is no reliable way to distinguish file and non-file properties.
closed 2024-10-24 by ra3xdh