ra3xdh/qucs_s #953: XYCE question and issue

tomhajjar, 2024-09-20:
1) I'm trying to understand how "BJT_Beta_sweep_mod_xyce.sch" netlist works in XYCE when it doesn't use the ".STEP" function. No XYCE netlist I can find in the Xyce_Regression/Netlists uses the method below. 

netlist:
I1 0  _net0 DC 0.5mA
Q1 _net1  _net0  0 2N2222A
V1 _net2  0 DC 12
VIc _net2 _net1 DC 0
.MODEL 2N2222A NPN (IS=3.0611E-14 .....)
dc 2n2222a:bf 200 250 12.5
.END

2) I then converted an ngspice Parameter sweep example "BJT_R_param_swp_xyce.sch" to see those netlists but can't get Trans to work even if I disable AC analysis. The AC analysis netlist does use .STEP as expected.

[image]
[image]

[2N2222_Amplifiers_xyce_prj.zip](https://github.com/user-attachments/files/17069100/2N2222_Amplifiers_xyce_prj.zip)

ra3xdh, 2024-09-20:
> I'm trying to understand how "BJT_Beta_sweep_mod_xyce.sch" netlist works in XYCE when it doesn't use the ".STEP" 

STEP command is not required for this schematic. The DC+Par.Sweep is converted to the DC sweep (DC command). Everything is correct here. 

> but can't get Trans to work even if I disable AC analysis. 

This is a well know problem existing since Qt4 version. Every sweep in the ParSweep+TRAN has different number of points. Qucs plotting engine can only process a stack of plots from Par.Sweep with the same points numbers. The AC is not affected by this.  The Ngspice uses `option interp` to make the plots uniform, but I don't know the similar command for Xyce. This cannot be fixed without plotting system redesign. I don't consider this to be done in the near future.

tomhajjar, 2024-09-20:
Any reason disabling the .TRAN simulation doesn't work? You have to delete it from the schematic or you get an Error.

[image]
[image]

ra3xdh, 2024-09-20:
> Any reason disabling the .TRAN simulation doesn't work? 

This is a bug. It puts the `.STEP` without attached simulation in the netlists.

tomhajjar, 2024-09-20:
Lastly to get .TRAN to work without an Error, you must delete BOTH the .TRAN parameter sweep AND the AC analysis. Disabling them doesn't work. Leaving the AC parameter sweep as-is doesn't bother it even though AC analysis in missing.

[image]

tvrusso, 2024-09-21:
> The Ngspice uses `option interp` to make the plots uniform, but I don't know the similar command for Xyce. This cannot > be fixed without plotting system redesign. I don't consider this to be done in the near future.

Xyce by default outputs every single time step that it computes.  However, this can be overriden using the `.options output` directive.    This can be used to tell Xyce to output at specific intervals (with interpolation so that the exact time values are output), or even have a schedule of output intervals so that only output in a particularly interesting time span is output in detail.

The simplest version would be something like:
```
.options output initial_interval=5us
```
to force output at precise 5 microsecond intervals, irrespective of the timestepping Xyce chooses in order to satisfy tolerances, with that interval being used for the entire simulation.   If Xyce's timestep doesn't hit that interval exactly, the output is interpolated using an interpolation order that matches the integration order.

If additional time/interval pairs are given after the `initial_interval=`, then different output intervals can be chosen for different parts of the simulation.

This is documented in section 2.1.22.8 of the Xyce Reference Guide version 7.8, on page 84.

To use this in qucs-s would require computing the actual desired output interval, but since you know the duration of the transient run and the number of points desired, that shouldn't be too difficult, and then you could probably have the plot work without having to redesign the plotting system.

tvrusso, 2024-09-21:
As for the `.dc 2n2222a:bf 200 250 12.5` question:   While `.DC` is normally used to perform a DC sweep over some voltage or current source, in Xyce it isn't actually much different from a .STEP, and it is possible to do a "DC Sweep" over most other parameters.  

In this case, it's sweeping over the BF parameter of the 2N222A model card.   The big difference between `.STEP` and `.DC` is that `.STEP` *only* varies the parameter, and requires some other analysis statement to actually run a simulation, whereas `.DC` performs an DC operating point computation at each step.   Replacing this `.DC` with a `.STEP` plus a single-point `.DC` line that only runs VIc at the value 0 (`.DC VIc 0 0 1`) would accomplish the same thing with more typing.

Per the Xyce reference guide, section 2.1.3, on the `.DC` analysis:

>Calculates the operating point for the circuit for a range of values. Primarily, this capability is applied to
independent voltage sources, but it can also be applied to most device parameters.

tomhajjar, 2024-09-21:
@tvrusso 
Does a simple GUI for XYCE exist for running simulations and generating plots like ngspice? I use it often to identify issues with Qucs-S and ngspice.

[image]
[image]

tvrusso, 2024-09-21:
Sadly, no, there is no GUI for Xyce at all.  Never has been, and given staff limitations and growing technical pressures in other directions, might never be (unless some interested third party creates one themselves).  At one time the team had high hopes for making Qucs-s its official GUI, but those hopes were dashed by forces above the team's pay grade.

Most Xyce users at Sandia have always just used command line and are very comfortable with it.  The output formats Xyce produces are all targeted at the plotting utilities various Xyce developers use --- gnuplot or tecplot for most developers, and in some cases at the tools Sandia circuit designers use (at the time of Xyce's early development most of our important users were using Cadence PSpice, which provides both schematic capture and plotting capabilities --- so Xyce is set up to output files that their plotting tool "probe" can use as well).

Xyce *runs* on Windows, but is primarily targeted at high-performance computational platforms running Unix-like operating systems.  On those systems, it's pretty easy to fire off Xyce from the command line and then invoke command line plotting tools.  It's not so easy on Windows, but the team is stretched so far (or at least was stretched so far before I retired) doing things that Sandia needs them to do that Windows doesn't get a whole lot of attention.  

The only thing I can suggest is that you get comfortable with Xyce's various output formats and pick one that works with a plotting utility you like (e.g. you could use Xyce's CSV output format and import the results into Excel).  The workflow would be kludgy, but if it's only to do side-by-side testing of Qucs-s and stand-alone, you shouldn't need to do it *too* often.

tvrusso, 2024-09-21:
It's also possible that the tools ngspice provides for use with rawfiles (like nutmeg) might be usable with Xyce's rawfile output.  (you can produce complete rawfile output with the `-r` command line option, with the argument after the `-r` being the name of the rawfile to dump).

ra3xdh, 2024-09-27:
Fixed by #965

PR #965 (2024-09-27): Fix disable simulation for XYCE
This PR fixes #953 The simulation disable for XYCE should work as expected.
closed 2024-09-27 by ra3xdh