ra3xdh/qucs_s #959: Simulations work in ngspice but not Qucs-S

tomhajjar, 2024-09-25:
I'm having an issue getting three Voltage Controlled Resistor models in the link below to work in Qucs-S. 
http://www.ecircuitcenter.com/Circuits/vc_resistor1/vc_resistor1.htm

ngspice:
I made three netlists, VCResistor1.cir, VCResistor2.cir and VCResistor3.cir and confirmed they work in ngspice.

Qucs-S:
VCRES2.cir is the same as the original model VCRES2.lib but uses .PARAM R=10k. I cannot get "VRES2_tran.sch" or "VRES2_lib_tran.sch" to work. 

[image]
[image]

[VCR_prj.zip](https://github.com/user-attachments/files/17136023/VCR_prj.zip)

ra3xdh, 2024-09-25:
The SPICE model is not implemented for VCresistor. It is not a standard SPICE device. It is represented as the equivalent subcircuit in the attached SPICE netlists. Furthermore this device is marked as SPICE-incompatible in Qucs-S, one cannot insert in SPICE mode. It never worked with SPICE backend. I can try to add SPICE model for this device in the future version.

tomhajjar, 2024-09-25:
You misunderstand, This project doesn't use VCresistor. It uses all Spice compatible components.

ra3xdh, 2024-09-25:
Ngspice netlist and Qucs-S schematic uses different VCresistor model. Try setting the low voltage of square wave source to 0.1V and the simulation will work. Otherwise you have division by zero inside the subcircuit equation.  It is not Qucs-S error. I can reproduce the same error with Ngspice CLI version after switching the model and setting the low voltage of the PWL source in the netlist to 0.0V.

tomhajjar, 2024-09-25:
Did you open the attached project? There is no "VCresistor". The input is a triangle wave.

[VCR_prj.zip](https://github.com/user-attachments/files/17136981/VCR_prj.zip)

ra3xdh, 2024-09-25:
Yes, I tried the simulation. Everything works after setting U0=0.1V on the source. Otherwise you have division by zero in the model. 

[image]

tomhajjar, 2024-09-25:
I updated all the files. The Author recommends using VCRES2. I didn't put extra components like 0.001 Volts/Ohms to stop divide by zero. A little more work and we might have a viable VCR for the library. 

There are numerous other VCR models available I didn't test.

[VCR_prj.zip](https://github.com/user-attachments/files/17137877/VCR_prj.zip)

tomhajjar, 2024-09-26:
New device should have some protection against "crashing" ngspice or qucsator-rf. Adding some 0.001 resistors or 1mv sources could help.

Another issue is the question of the "resistor" being polarity sensitive. I haven't tested this however. If this is an issue, the new symbol I created should reflect this.

ngspice has a  "Behavioral" resistor. This could be an alternative.

[image]

ra3xdh, 2024-10-23:
Implemented by #1016

PR #1016 (2024-10-23): Add SPICE entry for VCresitor
This PR fixes #959

* Added SPICE entry for VCRES. This device is represented by behavioral resistor
* Qucsator device made SPICE-independent and represented by equivalent EDD
* VCRES device moved from *verilog-a* category to *nonlinear devices*

[image]
[image]
closed 2024-10-23 by ra3xdh