ra3xdh/qucs_s #962: QUCS-S not returning data when running with ngspice

yuyangyumi, 2024-09-27:
When I run the following command in the command line, I cannot generate the s_param_wideband_amp.dat file.
```shell
qucs-s -n -i s_param_wideband_amp.sch -o s_param_wideband_amp.dat.ngspice --ngspice --run
```
error
```
Critical: "Schematic loading error! Unknown device R" (D:\a\qucs_s\qucs_s\qucs\components\component.cpp:1809, Component* getComponentFromName(QString&, Schematic*))
Error: Could not load schematic C:\project\person\java\omuao\test\s_param_wideband_amp.sch
```

yuyangyumi, 2024-09-27:
QUCS-S produces NGSpice Netlist without --run, but fails to return data with --run parameter.
How can I correctly obtain a .dat file via the command line?

ra3xdh, 2024-09-27:
It seems this CLI option was broken after simulator switching introduction at v1.1.0 I remember we did some patch fixing execution from CLI, but the simulator execution probably was not tested. Please wait for fix. There is no workaround.

ra3xdh, 2024-09-27:
Similar to #639 and #642

ra3xdh, 2024-09-27:
Fixed by #963. The fix will be available since the next release. You can also compile the application from source or use continuous build packages.

PR #963 (2024-09-27): Fix Ngspice run in CLI mode
Fixes #962
closed 2024-09-27 by ra3xdh