ra3xdh/qucs_s #967: qucsator-rf has issues versus qucsator

tomhajjar, 2024-09-28:
Having an issue with old qucs projects that used to work in qucs-s not working anymore. I am having the issue with many projects.

For this project, the AD822 spice model that worked in qucs AND qucs-s, doesn't work anymore in qucs-s.

[image]
[image]
[image]

[Opamp_AC_Tran.zip](https://github.com/user-attachments/files/17174199/Opamp_AC_Tran.zip)

zergud, 2024-09-28:
attached project work for me in current branch
[image]

tomhajjar, 2024-09-28:
Plots are blank for me

[image]

ra3xdh, 2024-09-28:
This issue is not related to Qucs-S GUI. It appeared after some recent modifications in qucsconv. Closing this. See https://github.com/ra3xdh/qucsator_rf/issues/26 to track the status. 

@zergud , You probably have some older version of qucsator/qucsconv. The v1.0.0 works as expected.