ra3xdh/qucs_s #968: Unable to modify simulation properties since version 24.3.2

njascgil, 2024-09-28:
I am unable to modify simulation properties in the latest 24.3.2 version and also in the continuous build (24.3.99 from less than a day ago). It still works correctly in the previous 24.3.1 version.
More specifically, I am unable to change settings such as Integration Method or its associated "display in schematic", but settings in other pages appear to work just fine (when these other pages exist, such as the sweep parameters for the transient simulation).

ra3xdh, 2024-09-28:
I confirm this bug. The bug is critical and I will prepare a release containing the fix in the next week.

ra3xdh, 2024-09-28:
Fixed by #969. Please either wait for the next release or use the continuous build. There is no workaround for the 24.3.2 version.

I  would also remember that the most of TRAN simulation settings like "Integration method" have no effect in SPICE mode. This is another known issue. It's need to provide a way to show/hide simulator-incompatible properties. The solution is not straightforward and beyond the scope of this issue.

closed 2024-09-28 by ra3xdh

njascgil, 2024-09-28:
Thank you so much! 👍 (I still use mostly Qucsator, so I didn't even noticed the SPICE mode issue...)

ra3xdh, 2024-09-28:
> I still use mostly Qucsator, so I didn't even noticed the SPICE mode issue

Also take into account very poor time-domain performance of the Qucsator that is still be inherited by QucsatorRF. It is not recommended unless you are using RF devices like microstrips.

njascgil, 2024-09-28:
Thanks! I'll keep that in mind. However, as I am currently doing mainly some academic analysis of low frequency power converters (using transformers, thyristors, diodes, TRIACs and so on) Qucsator seems to be sufficient and also a bit more convenient/lenient when considering both the default models and also the voltage/current ranges we usually test.

PR #969 (2024-09-28): Fix simulation properties edition
This PR fixes #968