ra3xdh/qucs_s #970: MOSFET Libraries qucsator issues

tomhajjar, 2024-09-28:
I was trying to convert some ngspice audio MOSFET power amplifier projects over to work with qucsator. I can get AC and DC simulations to work but Trans does not.

I then tried using MOSFETs from the built in "Qucs" Libraries in simple Tran circuits and cannot get any device to work with qucsator even after removing the inductor and capacitor. Are the MOSFET Libraries marginal under qucsator?

[image]
[image]

[MOSFET_test_qucs_prj.zip](https://github.com/user-attachments/files/17175655/MOSFET_test_qucs_prj.zip)
[Amp_100W_qucs_prj.zip](https://github.com/user-attachments/files/17175667/Amp_100W_qucs_prj.zip)

ra3xdh, 2024-09-28:
Qucsator has very poor time-domain performance. It is not Qucs-S GUI issue. See https://github.com/Qucs/qucsator/issues/19 Switch simulaiton may fail even without LC elements. The MOSFET device in the library contains the parasitic elements inside which may cause convergence issues with Qucsator.  The fixing of this is not a straightforward task. The help from simulation algorithms experts is highly appreciated.

closed 2024-09-28 by ra3xdh

ra3xdh, 2024-09-28:
The MOSFET libraries and qucsator interaction from the GUI side remains untouched since the 0.0.19 version. Also I didn't change the TRAN computation algorithm in the QucsatorRF. All known TRAN simulation issues of Qucsator still be inherited by QucsatorRF.