ra3xdh/qucs_s #972: No OP analysis results

dwarning, 2024-09-29:
Any of my testbenches show 0 voltages and currents after pressing F8 key.
[image]

From what is it dependent? The spice4qucs.cir is written under ~/.qucs/spice4qucs but I see no spice4qucs.cir.dc_op result file. The cir file has not included the pre_osdi file as shown in the schematic and can't simulated.

ra3xdh, 2024-09-29:
I cannot reproduce this issue. Everything works as expected on my machine. Please submit your schematic including external Verilog-A source/OSDI files. 

The most probably the Nutmeg script is working wrong and model is not loaded. Every nutmeg script is a separate simulation and cannot add the code before the DC sweep. I am wondering if this worked before.  You have to put this script in SpiceInit at *SPICE init sections->SPICE init*. 

> but I see no spice4qucs.cir.dc_op result file. 

The datasets are moved to `.cache` location. See #145 The default is `$HOME/.cache/qucs-s`

[image]

ra3xdh, 2024-09-29:
Also the *INCLUDE script* device may help with the following contents:
~~~
.control
pre_osdi file.osdi
.endc
~~~

dwarning, 2024-09-29:
Yes substitute nutmeg script with .spiceinit helps for OP analysis and other simulations too.
[image]
Thank you!

closed 2024-09-29 by dwarning