ra3xdh/qucs_s #973: Not able to open files from the command line

njascgil, 2024-09-29:
Using the "-i" parameter I am not able to open files from the command line. Qucs-s opens fine, but with the default empty document.
At first I though it might have something to do with spaces in the path, but it does not work even if I open a cmd prompt inside the qucs-s.exe folder and copy the sch file also to this location (then trying "qucs-s.exe -i pl1.sch").
Is this a real issue or am I doing something wrong?

ra3xdh, 2024-09-29:
I cannot reproduce this. The `qucs-s -i schematic.sch` opens a document on my machine. Probably you may have $PATH issues on Windows. Try to specify the full path to file.

njascgil, 2024-09-29:
I think I may have found the issue, but perhaps it's by design.
When the default simulator is Qucsator (e.g. by closing Qucs-s with that simulator selected in the toolbar), opening a sch via the command line works just fine. If the default is Ngspice instead, Qucs-s starts with the default "untitled" empty sheet.
(I found it after running Qucs-s inside a VM and comparing the registry entries for both cases.  `DefaultSimulator=0x08` is ok, but `DefaultSimulator=0x01` is not...)

ra3xdh, 2024-09-29:
Sorry, I still cannot reproduce this. The Qucs-S opens schematic from CLI using both Qucsator and Ngspice modes.

ra3xdh, 2024-09-29:
The exception is when schematic contains SPICE-incompatible or Qucsator-incompatible device. It shows an empty schematic and prints in the console message like this:
~~~
Critical: "Schematic loading error! Unknown device K_SPICE"
~~~

njascgil, 2024-09-29:
That may be it. When I remove the (old) Qucs transformer from my model the sch opens just fine either way.
However, I do not see that message when trying to open the sch containing the Spice incompatible file...

ra3xdh, 2024-09-29:
Maybe Windows redirects the `stderr` stream and you don't see the error messages in CLI.

ra3xdh, 2024-09-29:
I have added TODO list for this issue. The first task is to allow to load schemtic with incompatible devices from CLI. The CLI options to load schematic needs a bit redesign. The `-i` option should not be required. The syntax should be `qucs-s schematic.sch` as used by all applications. This will allow to associate SCH files to Qucs-S by system tools.

closed 2026-10-17 by ra3xdh