ra3xdh/qucs_s #974: Refactor Property class

ra3xdh, 2024-09-29:
The `Property` is inherited mostly unchanged from Qt3 application version and may benefit from the refactoring. I propose the following things:

* Add a field containing property type: file, equation or usual property
* Add a field containing allowed simulator type: SPICE, Qucsator, etc.

The implementation of the second will allow to hide simulator-incompatible device properties from user to prevent its accidental setting and usage.

ra3xdh, 2025-01-06:
Transferred to the next release cycle.

wawuwo, 2025-05-02:
>Add a field containing allowed simulator type: SPICE, Qucsator, etc.

@ra3xdh are the members of `spicecompat::Simulator` meant here?

wawuwo, 2025-05-07:
@ra3xdh could please take a look at previous comment?

ra3xdh, 2025-05-07:
> are the members of spicecompat::Simulator meant here?

Yes, it this enum. The logic should be following. If the property is not simulator compatible hide it from component properties dialog and schematic view. Otherwise it may confuse user. The default value should `simAll`

ra3xdh, 2025-05-07:
> Add a field containing property type: file, equation or usual property

The idea behind this is to get rid of hack with "Symbol" property name like in `param_sweep.cpp` line 68 and other places.

closed 2025-05-15 by ra3xdh