ra3xdh/qucs_s #976: QucsStudio file converted to Qucs crashes Qucs-S

tomhajjar, 2024-09-29:
I realize this is an odd thing to complain about but a QucsStudio file converted to Qucs and opens in Qucs crashes Qucs-S. I figured the file crash issue could be important. Normally Qucs would flag the file as bad. 

After I hand edit a QucsStudio file to "Qucs" format I then use Qucs to fix other compatibility issues. Qucs also "cleans" up the file when it is saved. I have converted many QucsStudio files to Qucs. It wasn't that hard up to QucsStudio 2.5.7 but the QucsStudio 4.3.1 file format has made it 10x as hard. Qucsconv in 4.3.1 works better than 2.5.7 so that is how I ended up using 4.3.1. 

I was trying to figure out how the MOSFET Library worked in QucsStudio. If one looks at IRFZ44N in J_MOSFETs.lib, it is almost identical to the manufacturers model IRFZ44N.spi. I was trying to figure out how do get it to work in qucsator.

[image]

[MOSFET_test_qucs_prj.zip](https://github.com/user-attachments/files/17180176/MOSFET_test_qucs_prj.zip)

ra3xdh, 2024-09-29:
> QucsStudio file converted to Qucs and opens in Qucs crashes Qucs-S.

The reason of the crash is wrong capacitor line format. Correct line as following:

~~~
<C CAP 1 150 330 10 15 0 0 "2.06741e-009" 0 "0" 0 "neutral" 0>
~~~

It seems the QucsStudio doesn't have the Symbol property of the capacitor. Such line formatting may appear only from schematics converted from QucsStudio. This error should not appear while the normal operation of Qucs-S and usage of normal Qucsconv utility. The schematics converted from QucsStudio are in general not supported by Qucs-S. I have just checked if Qucs-S opens the files from Qucs-S v0.0.5 and everything worked normally. So, this issue is only related to files obtained from QucsStudio converter and may be ignored. 

> I was trying to figure out how the MOSFET Library worked in QucsStudio.

The IRFZ44 model looks the same as used in NMOSFETs.lib for Qucs-S and Qucs. The Git history says this library remains unchanged since 2008. QucsStudio uses another computation algorithm for TRAN analysis. This may explain why this model works for QucsStudio simulator, but doens't work for Qucsaror/QucsatorRF. I cannot backport this computation algorithm because the QucsStudio is closed source software. 

> It wasn't that hard up to QucsStudio 2.5.7 but the QucsStudio 4.3.1 file format has made it 10x as hard. Qucsconv in 4.3.1 works better than 2.5.7

I have no wish to add support for QucsStudio files conversion or to accept schematic files converted by QucsStudio files converter utility. QucsStudio is closed source and Windows-only software and I have no intention to help its developer. Please understand me correctly, but when you are trying to import QucsStudio schematics in Qucs-S, you are on your own.

closed 2024-09-29 by ra3xdh

tomhajjar, 2024-09-29:
I only mentioned it because of the crash. Wasn't sure if this could cause issues in the future.