ra3xdh/qucs_s #979: Crash in saveApplSettings

papoteur-mga, 2024-09-30:
I just build qucs-s with Qt6.4 in Linux
I launch it, then apply the default values in the first window.
I get this crash:
```

Thread 1 "qucs-s" received signal SIGSEGV, Segmentation fault.
0x00007ffff77fa879 in QWidget::screen (this=0x0) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qwidget.cpp:2506
2506        if (auto associatedScreen = d->associatedScreen())
(gdb) bt full
#0  0x00007ffff77fa879 in QWidget::screen (this=0x0) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qwidget.cpp:2506
        associatedScreen = <optimized out>
        d = <optimized out>
        topLevel = <optimized out>
        topData = <optimized out>
        screenByPos = <optimized out>
#1  0x00007ffff77fa985 in QWidget::saveGeometry (this=0x0) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qwidget.cpp:7348
        array = {d = {d = 0x0, ptr = 0x0, size = 0}, static _empty = 0 '\000'}
        stream = {<QIODeviceBase> = {<No data fields>}, d = {d = 0x0}, dev = 0x2b96b70, owndev = true, noswap = false, byteorder = QDataStream::BigEndian, ver = 7, 
          q_status = QDataStream::Ok}
        magicNumber = 31051979
        majorVersion = 3
        minorVersion = 0
        screenNumber = <optimized out>
#2  0x00000000004e97d0 in saveApplSettings() ()
No symbol table info available.
#3  0x00007ffff65b96d5 in void doActivate<false>(QObject*, int, void**) () from /usr/lib64/libQt6Core.so.6
No symbol table info available.
#4  0x00007ffff78d3f22 in QAbstractButton::clicked (this=this@entry=0x2b6ac20, _t1=<optimized out>)
    at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/build/src/widgets/Widgets_autogen/include/moc_qabstractbutton.cpp:368
        _a = {0x0, 0x7fffffffc38c}
#5  0x00007ffff78d414a in QAbstractButtonPrivate::emitClicked (this=0x2b789d0) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/widgets/qabstractbutton.cpp:379
        q = 0x2b6ac20
        guard = <optimized out>
#6  0x00007ffff78d56b8 in QAbstractButtonPrivate::click (this=0x2b789d0) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/widgets/qabstractbutton.cpp:372
        q = <optimized out>
        changeState = <optimized out>
        guard = <optimized out>
#7  0x00007ffff78d58c6 in QAbstractButton::mouseReleaseEvent (this=0x2b6ac20, e=0x7fffffffc980) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/widgets/qabstractbutton.cpp:973
        d = 0x2b789d0
#8  0x00007ffff780bc08 in QWidget::event (this=0x2b6ac20, event=0x7fffffffc980) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qwidget.cpp:9205
        d = <optimized out>
#9  0x00007ffff77b6261 in QApplicationPrivate::notify_helper (this=this@entry=0xb6cba0, receiver=receiver@entry=0x2b6ac20, e=e@entry=0x7fffffffc980)
    at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qapplication.cpp:3315
        consumed = false
        filtered = false
#10 0x00007ffff77bf9c7 in QApplication::notify (this=<optimized out>, receiver=0x2b6ac20, e=0x7fffffffc980)
    at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qapplication.cpp:2803
        me = <incomplete type>
        mouse = 0x7fffffffc980
        relpos = {xp = 62, yp = 10}
        eventAccepted = <optimized out>
        pw = {wp = {d = 0x2b18190, value = 0x2b6ac20}}
        w = 0x2b6ac20
        d = <optimized out>
        isWindowType = <optimized out>
        isWidgetType = <optimized out>
        res = false
--Type <RET> for more, q to quit, c to continue without paging--c
#11 0x00007ffff655b798 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib64/libQt6Core.so.6
No symbol table info available.
#12 0x00007ffff77be2c4 in QApplicationPrivate::sendMouseEvent (receiver=receiver@entry=0x2b6ac20, event=event@entry=0x7fffffffc980, alienWidget=<optimized out>, alienWidget@entry=0x2b6ac20, nativeWidget=0x2b5a290, buttonDown=buttonDown@entry=0x7ffff7d78a90 <qt_button_down>, lastMouseReceiver=..., spontaneous=true, onlyDispatchEnterLeave=false) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qapplication.cpp:2387                                                                                                                                       
        receiverGuard = <optimized out>
        nativeGuard = <optimized out>
        alienGuard = <optimized out>
        activePopupWidget = <optimized out>
        graphicsWidget = false
        widgetUnderMouse = <optimized out>
        wasLeaveAfterRelease = <optimized out>
        result = true
#13 0x00007ffff781d1b7 in QWidgetWindow::handleMouseEvent (this=0x2b96750, event=0x7fffffffcc20) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qwidgetwindow.cpp:623
        translated = <incomplete type>
        contextMenuTrigger = QEvent::MouseButtonPress
        widget = 0x2b6ac20
        mapped = {xp = 62, yp = 10}
        initialPress = <optimized out>
        receiver = 0x2b6ac20
#14 0x00007ffff781fc68 in QWidgetWindow::event (this=0x2b96750, event=0x7fffffffcc20) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qwidgetwindow.cpp:241
No locals.
#15 0x00007ffff77b6261 in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x2b96750, e=0x7fffffffcc20) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/widgets/kernel/qapplication.cpp:3315                                                                                                                                                                          
        consumed = false
        filtered = false
#16 0x00007ffff655b798 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib64/libQt6Core.so.6
No symbol table info available.
#17 0x00007ffff6deeda3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib64/libQt6Gui.so.6
No symbol table info available.
#18 0x00007ffff6e3ee7c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt6Gui.so.6
No symbol table info available.
#19 0x00007ffff4f3204a in xcbSourceDispatch(_GSource*, int (*)(void*), void*) () from /usr/lib64/libQt6XcbQpa.so.6
No symbol table info available.
#20 0x00007ffff5f17689 in g_main_context_dispatch () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#21 0x00007ffff5f17a18 in g_main_context_iterate.constprop () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#22 0x00007ffff5f17a9c in g_main_context_iteration () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#23 0x00007ffff67f59ec in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt6Core.so.6
No symbol table info available.
#24 0x00007ffff65671c3 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt6Core.so.6
No symbol table info available.
#25 0x00007ffff79f99f7 in QDialog::exec (this=0x2b5a290) at /usr/src/debug/qtbase6-6.4.1-5.mga9.x86_64/src/corelib/global/qflags.h:74
        eventLoop = <incomplete type>
        d = 0x2782610
        deleteOnClose = false
        wasShowModal = false
        guard = <optimized out>
        res = <optimized out>
#26 0x00000000005068f9 in QucsApp::slotSimSettings() ()
No symbol table info available.
#27 0x0000000000518880 in QucsApp::QucsApp() ()
No symbol table info available.
#28 0x00000000004df43b in main ()
```
I don't know if the problem is with your application, or Qt 6.
Papoteur

papoteur-mga, 2024-09-30:
A similar error occurs with Qt5
```
(gdb) bt full
#0  0x00007ffff798ecb4 in QWidget::frameGeometry() const () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#1  0x00007ffff798edb3 in QWidget::saveGeometry() const () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#2  0x00000000004cdac1 in saveApplSettings() ()
No symbol table info available.
#3  0x00007ffff6cd3f84 in void doActivate<false>(QObject*, int, void**) () from /usr/lib64/libQt5Core.so.5
No symbol table info available.
#4  0x00007ffff7a4d652 in QAbstractButton::clicked(bool) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#5  0x00007ffff7a4d86a in QAbstractButtonPrivate::emitClicked() () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#6  0x00007ffff7a4f2f8 in QAbstractButtonPrivate::click() () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#7  0x00007ffff7a4f507 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#8  0x00007ffff79a0d28 in QWidget::event(QEvent*) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#9  0x00007ffff796240e in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#10 0x00007ffff796a2c2 in QApplication::notify(QObject*, QEvent*) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#11 0x00007ffff6c9e888 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib64/libQt5Core.so.5
No symbol table info available.
#12 0x00007ffff796844b in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#13 0x00007ffff79b9059 in QWidgetWindow::handleMouseEvent(QMouseEvent*) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#14 0x00007ffff79bc1b9 in QWidgetWindow::event(QEvent*) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#15 0x00007ffff796240e in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#16 0x00007ffff6c9e888 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib64/libQt5Core.so.5
No symbol table info available.
#17 0x00007ffff713c13c in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib64/libQt5Gui.so.5
No symbol table info available.
#18 0x00007ffff7110f2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Gui.so.5
No symbol table info available.
#19 0x00007ffff371cb8a in xcbSourceDispatch(_GSource*, int (*)(void*), void*) () from /usr/lib64/libQt5XcbQpa.so.5
No symbol table info available.
#20 0x00007ffff5c4c689 in g_main_context_dispatch () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#21 0x00007ffff5c4ca18 in g_main_context_iterate.constprop () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#22 0x00007ffff5c4ca9c in g_main_context_iteration () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#23 0x00007ffff6cf31f6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
No symbol table info available.
--Type <RET> for more, q to quit, c to continue without paging--c
#24 0x00007ffff6c9d46b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib64/libQt5Core.so.5
No symbol table info available.
#25 0x00007ffff7b53247 in QDialog::exec() () from /usr/lib64/libQt5Widgets.so.5
No symbol table info available.
#26 0x00000000004e9ca9 in QucsApp::slotSimSettings() ()
No symbol table info available.
#27 0x00000000004f773f in QucsApp::QucsApp() ()
No symbol table info available.
#28 0x00000000004c4919 in main ()
```

ra3xdh, 2024-09-30:
I have just tested a fresh start of the application and I cannot reproduce this. There is no crash using Manjaro+Wayland+Qt6.7.2 Please post a full information about your platform. A test in the VM environment may be required.

papoteur-mga, 2024-09-30:
My system is Mageia 9, using Plasma - X11
Graphic driver is amdgpu.
Do you need other info?

ra3xdh, 2024-10-01:
> My system is Mageia 9, using Plasma - X11

Yes, this information is sufficient.

I have also tried fresh application start using Ubuntu+Qt6.2.4 and still cannot reproduce the crash. The test in VM using Mageia is required.

papoteur-mga, 2024-10-10:
> > My system is Mageia 9, using Plasma - X11
> 
> The test in VM using Mageia is required.

I just did one, in Mageia Cauldron (development version). I still get a crash as exposed above.
Qt6.7

ra3xdh, 2024-10-11:
I have installed Mageia9 in the VM and I can reproduce the crash. The Qucs-S crashes only if it cannot find any simulator on the first startup. This should not normally happen. The Qucs-S should find at least QucsatorRF with is compailed together with GUI. 

I have found the following issues in Mageia installation:

* You probably forgot `git submodule init && git submodule update` after the clone repository. Qucsator was not compiled and installed. Please try to recompile Qucs-S with submodules.
* The Mageia Ngspice package seems to be broken. It doesn't contain `ngspice` executable. Please compile Ngspice from source.

[image]

**Workaround for this issue**

Press the *Cancel* button in the dialog window on the first application start. You will be able to launch the application and configure simulators later using *Simulation->Simulator settings*. But anyway you should provide a correct Ngspcie installation and compile Qucs-S with submodule.

**TODO**

I will provide a fix for crash soon. Anyway the application should not crash if the simulator not found by some reason.

ra3xdh, 2024-10-11:
The crash fixed by #992

closed 2024-10-11 by ra3xdh