ra3xdh/qucs_s #980: Replacement for Qucs Verilog-A Library

tomhajjar, 2024-10-01:
Slowly making replacements for the Qucs Verilog-A Library devices.

1) Modded VCR.sch to get rid of divide by zero if Vin=0 volts. What actual VCR is chosen is TBD.
2) A new mod-amp was created using a Spice model supplied by Holger, @holvo
3) Qucs Curtice EDD based MESFET has been updated by @ra3xdh to work in ngspice.

Other devices will take time. The Log amp should be easy to do.

[image]

[image]
[image]
[image]

[Verilog-A_Lib_qucs_prj.zip](https://github.com/user-attachments/files/17213627/Verilog-A_Lib_qucs_prj.zip)

tomhajjar, 2024-10-03:
Made a Voltage Controlled Amplifier (VCA) from the Ideal Multiplier.
Presently the Control Voltage, Vc is not bounded. The voltage can be any positive or negative voltage. Gain is negative when Vc <0V. 

[image]
[image]

[VCA_prj.zip](https://github.com/user-attachments/files/17238697/VCA_prj.zip)

ra3xdh, 2024-10-03:
The replacement of the Qucsator Verilog-A devices for Ngspice is not straightforward task. The SPICE entries may be directly generated only for VCRES and POT devices. The firs in fact VCCS and the second is two series resistors. Other devices in principle may be integrated only as subcircuit macromodel. Qucsator doesn't support something like SPICE `.INCLUDE` directive. Only integration via library is possible. 

The current implementation doesn't support attaching subcircuits to to hardcoded C++ system library devices. The extension at C++ level is needed here. The XML library may a transitional solution. 

The better solution would be implementing the #343 Maybe I will return to this task after the status of the OpenVAF will become clear.

Summarizing all above, I would keep the *Verilog-A devices* group except VCRES and POT as is.  These devices are rarely used and were designed as a demo of Qucs+ADMS link. Let's wait for implementation of #343. The QucsatorRF should be switched to OSDI interface in the future. But I don't consider implementation of this feature in the near future because of short of time.

closed 2024-12-02 by ra3xdh