ra3xdh/qucs_s #982: Different Data for three "identical" circuits using Qucsator

tomhajjar, 2024-10-04:
I ran into an issue testing an old Qucs-S project that was previously converted from Qucs. First issue was the 4-pin Blue BJTs were removed from Qucs-S. I then converted to 3-pin Blue but then the "old" Nutmeg equations didn't work because of other changes to Qucs-S. It wasn't obvious what changes were needed to the Nutmeg equations so I went back to Qucs and the original project.

This is when I found I couldn't get "Rout" for "mirrorwilson.sch" in different project folders to be the same. I renamed each file and placed them in the same folder "Mirrors_qucs_prj". I confirmed the data for each version of "mirrorwilson.sch" was different when tested in Qucs and Qucs-S so qucsator/qucsator-rf wasn't the issue. 

1) Can you spot why the seemingly identical files have different Rout?
2) I have to redo the Nutmeg equations to see what Rout will be under ngspice. My old data is gone.

[image]
[image]
[image]
[image]

[Mirrors.zip](https://github.com/user-attachments/files/17260113/Mirrors.zip)

ra3xdh, 2024-10-04:
> First issue was the 4-pin Blue BJTs were removed from Qucs-S. 

It is not removed. It is hidded for Ngspice by mistake. I will revert this device in the next patch. 

> Can you spot why the seemingly identical files have different Rout? I didn't see anything in a text editor

It is not a bug. The result form Qucsator and SPICE may be different. You have ideal BJT model. It acts as the ideal current source and you have `dI=0` in the `R=dV/dI` equation. The `dI=0` may give an unstable result and random noise when approaches to machine epsilon. The SPICE model may act differently and give different result. Set the `VAF`parameter of the BJT or use the real SPICE model instead of ideal and get the different result. The same is valid for the initial part of the `Rout` curve. The BJT again acts as the ideal source. 

[image]

> I have to redo the Nutmeg equations to see what Rout will be under the latest Qucs-S/ngspice. 

It is again not a bug. You have to specify the simulation name instead of `dc` in the Nutmeg equation. The `DC` is a valid simulation name for DC OP analysis. See #463, #513, and #124. You have to specify the `SW1` to attache the equation to DC OP. After making these corrections everything works as expected. 

[image]

ra3xdh, 2024-10-04:
The PR #983 makes BJTsub device available.

closed 2024-10-04 by ra3xdh

tomhajjar, 2024-10-05:
Updated ngspice and qucs versions of the project. Might be useful for someone.

[Currrent_Mirrors.zip](https://github.com/user-attachments/files/17264741/Currrent_Mirrors.zip)

ra3xdh, 2024-10-05:
The SPICE behavior depends on GMIN setting. The default `GMIN=1e-12` works fine, but random noise starts from `GMIN=1e-14`. The `GMIN=1e-16` fails to converge. The Qucsator have another behavior. I didn't find GMIN setting for Qucsator DC. 

[image]

tomhajjar, 2024-10-05:
https://qucs.sourceforge.net/tech/node29.html#SECTION00762200000000000000

PR #983 (2024-10-04): Fix of BJTsub and Diode device
This PR contains the following improvements:

* Make BJTsub available for all simulators; Fixes #982 
* Fix `UseGlobTemp` parameter name typo in diode
* Allow fill from SPICE for BJTsub