ra3xdh/qucs_s #985: Qt6 problem

dwarning, 2024-10-06:
I made no change in my qt6 installation under Ubuntu 22.04 since long time.

Building qucs-s from _current_ branch with
cmake .. -DWITH_QT6=ON -DCMAKE_INSTALL_PREFIX=~/.local

I got 
-- Could NOT find Qt6Charts (missing: Qt6Charts_DIR)
CMake Error at qucs-s-spar-viewer/CMakeLists.txt:35 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake

  but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find Qt component "Charts".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6Charts/Qt6ChartsConfig.cmake" does NOT
  exist

That's new - building the program (24.3.1) with Qt6 enabled a month ago works w/o problem.

ra3xdh, 2024-10-06:
It is not build system error. The new S-parameter viewer utility requires Qt6Charts. Install the package `libqt6chart6-dev` with the package manager. It is not a part of default Qt6 setup. I have updated the REAME.md and build instructions on Wiki.

closed 2024-10-06 by ra3xdh

dwarning, 2024-10-06:
on Ubuntu 22.04
maybe a typo:
libqt6chart6-dev -> libqt6charts6-dev
Thank you