ra3xdh/qucs_s #1001: Qucs-S reports wrong ngspice log data - Issue getting complex Spice model that works in LTspice to work in Qucs-S

tomhajjar, 2024-10-18:
I have been trying to get a complex library of IGBT Spice models to work in Qucs. The Library works in LTspice as-is. 

The original lib file "IGBT_600V_HighSpeed3_L1.lib" was simplified to "IGP20N60H3.cir" hoping to get Qucs/ngspice to work. Unfortunately I get a conflicting "Qucs-S Message", Simulator crashed! and Simulator successful.

I then sent the netlist to ngspice via DuSpice and get a totally different "Message" in the ngspice file log, "IGP20N60H3_DC_sweep.log". 

1) Are the two "Formula() error." in the Log confusing Qucs-S?

2) What is the proper way to handle complex Spice Libraries like "IGBT_600V_HighSpeed3_L1.lib"? I'm sure my editing of the file to create "IGP20N60H3.cir" wasn't done 100%. 

[image]
[image]

[IGBT_prj.zip](https://github.com/user-attachments/files/17441449/IGBT_prj.zip)
[IGBT_LTspice.zip](https://github.com/user-attachments/files/17441448/IGBT_LTspice.zip)

ra3xdh, 2024-10-20:
> Would it be useful to allow the User to see both Qucs "Message" and ngspice "Log" Files?

Qucs-S doesn't apply any filters to log files and doesn't hide the log data from user. The problem is more complicated.  The problem may be in Qt itself. The `QProcess::readAllStandardOutput` returns an empty string if the child process is terminated with error code. And Qt cannot obtain the log if the Ngspice is terminated with error code. The simulation process has merged `stdout` and `stderr` channels. Need further investigation. I didn't understand what happens. Don't expect a quick fix. 

> What is the proper way to handle complex Spice Libraries like "IGBT_600V_HighSpeed3_L1.lib"? 

You have to extract all dependent parameters and child subcircuit in the single `*.cir` file. Also wait until #944 will be implemented.

closed 2024-11-12 by ra3xdh

tomhajjar, 2024-11-12:
This issue points to ngspice lacking the ability to process many of the "complex" Spice models that work in LTspice.

Some models work after editing, but I lack the knowledge to make the major edits required.

commit 3040c2cb (2024-11-12): Fix #1001
commit 94c2ca50 (2024-11-12): Merge pull request #1057 from ivandi69/1001_fix

Fix #1001