ra3xdh/qucs_s #997: Frozen application when changing signal in Cartesian Diagram

csrabak, 2024-10-18:
This strange bug happened in a friend's machine and after a lot of tests we cand reproduce it in my Windows 10 machine, Qucs-S 24.3.2:

[image]

After the diagram for the output voltage of the test circuit has been all set, the user "changes its mind" and decides to replace the voltage signal to the relative dB one given by the Nutmeg equation.

Steps to reproduce:

1. After sucessful simulation, on the diagram, select the output signal, change the X scale to log, and the Y one to log too.
2. Get back to the simulator canvas and see the results as the png attached.
3. Now, edit the diagram and select the 'y' (equation signal) and replace the one existing, i.e., add y, remove `ngspice/ac.v(vout)`.

The whole application freezes and the OS keeps "not responding" for the program, and the only resort is to stop it via Windows Task Manager.

It seem that the selection of Y log and changing of the signal vector confuses the program hard enough that it gets frozen.

I'm quite sure this is a behaviour should be considered as bug.

ra3xdh, 2024-10-18:
Yes, I confirm this problem. The `y` variable has negative value. This should not be allowed for logarithmic scale. It's need to implement a check for negative variable and show no plot. Also you have a wrong syntax in the equations. It should be `db(v(vout))` but not `db(vout)`. I am wondering that Ngspice accepts this.

csrabak, 2024-10-18:
@ra3xdh 
The Ngspice 43 manual gives the syntax for `db(<vector)` as equivalent  to `db(mag(<vector>))` where `mag(vector)` is "the same as `abs(<vector>)`", page 363.

As for ngspice the correct way of doing it is via Nutmeg equations, I've tried to use a more compatible as possible syntax, as it would be nice if the same circuit could be run in other Spice simulators Qucs-S supports.

Perhaps when we give a specific name to a node (as in the example Vin and Vout) the parser already defaults to `v(<node name>)` ?

tomhajjar, 2024-10-18:
Here is a project that will help you out. You can copy and paste the Nutmeg equations if you use in/out labels.

[image]

[AC_Tran_Four_FFT_prj.zip](https://github.com/user-attachments/files/17441672/AC_Tran_Four_FFT_prj.zip)

csrabak, 2024-10-19:
@tomhajjar,

Thanks for the example!

I think we have to pinpoint the issue in order we could be contributive in the discussion.

Vadim's comment pertains to the correctness of the way I wrote in my example, which see, **works**.

On the other hand, a search in the Ngspice current manual (ver. 43 as the date of this post) shows a mix of "styles" including the one used in my example and ones like the suggested by you two.

As a matter of fact, in Windows, using Vogt's utility DuSpiceStart and issuing in it `dB(Vout)` in the GUI happily plots the log voltage as [I would've] expected. . .

This means that in Tom's example, in the Nutmeg block the equation for the gain in dB can be written as `Gain_dB=dB(out)`, here going to the same level of simplification Tom done for the computation of the phase, as it is enthymematic that the starting phase of the driving voltage is zero and the value of the peak of the voltage is one volt, which divides all the Vout vector by one.

If the input voltage in the circuit under investigation could have voltages different or even variable, as in the input of a filter, etc., a more "complete" canonical, but still simple equation would be `Gain_dB=dB(out/in)` (also tested outside Qucs-S, so it is valid for Ngspice).

So, the matter in discussion is if we have a more correct or preferable way of referencing the vectors for printing, ploting and using it in the equations, or all the ways the program accepts are the result of compatibility with historic versions or with other Spice implementations?

PS.: I would like to keep the reasoning that whereas this is presently customary to express gain in this fashion, it is incorrect as the impedances in the input and output of the amplifier are very different, and in order the gain could be used as it was intended an additional term relating the impedances would have to be put in the equation.  I think we had a thread about this some years ago, and the strenght of tradition overwhelms the correctness when using decibels in general electronics.

tomhajjar, 2024-10-19:
An immense amount of work was done to get multiple analysis to work on the same schematic like AC, DC, Trans, FFT. 

The Vector name to be plotted can be different depending on what analysis is being done.

By coincidence, Qucs math functions and ngspice math functions in Nutmeg have a few function names that are identical. Some old Qucs equations work in ngspice but it is problematic because you could have an equation block where 4 out of 5 equations work but the one that fails causes the entire block to be rejected.

csrabak, 2024-10-19:
In reverse order:

I didn't include Qucs (legacy) stuff in my observation as I think is a moot discussion.

I do not conflate Power Gain with Voltage Gain, the equation **20***log10(V1/V2) does! It is a terminology problem that comes from the fifties of the last century and now is entrenched in our jargon, although mathematically and conceptually wrong.

I concur, and I congratulate all the effort Qucs-S does, that generating a netlist from the schematic capture is a complex task as both projects Qucs and Ngspice have their history, and the latter has to cope with compatibilities of Spice  2/3. Right now in our little team we're discussing if using "Meg" altogether in schematics supposed to be simulated is better than remembering to set the appropriate ngspice option elsewhere in the schematic¹. 

Yes, the differences may be an historic accident or being the work of a team which may have diverged a little, and Qucs-S needs to generate the acceptable for the range of supported versions of Ngspice in order to be useful.

Getting back to @ra3xdh comment, and your mention on the dependency of the type of analysis about the acceptable ways to call the vector names, I think we should not add restrictions on what Qucs-S exports, although I see a reasoning on the way the interface Qucs-S does for equations where one can choose one or all simulations, and I believe @tomhajjar's point is that it may break for some simulation.

Point noted and taken! Thanks. 

[1] In Qucs-S is less than an issue as we don't have facilities (and the problems that come from) to generate BOM or export to PCB program. If this landscape changes, then another direction could be more appropriate.

ra3xdh, 2024-10-19:
The infinite loop fixed by #1005

commit 631cf16f (2024-10-19): Merge pull request #1005 from ivandi69/997_fix

Fix #997
commit 80be7d55 (2024-10-19): Fix #997
closed 2024-10-19 by ra3xdh