ra3xdh/qucs_s #1002: Missing text in all labels or symbols

aisuneko, 2024-10-19:
All texts present in the schematics (except axis labels on diagrams) are not rendered properly, shown as blanks missing on the screen:
[image]

Clicking on and selecting text boxes in certain labels might reveal the text within, but they seem to be transparent and are not actually rendered on the screen. Right clicking on these hidden text labels might show a "fake" popup menu with its color the same as the schematic background but unable to interact with.
[image]

Terminal will be spammed with lots of `Warning: The requested buffer size is too big, ignoring (:0, )` messages when launched from the command line.
Qucs-S 24.3.0, Arch Linux, Qt 6.8.0

ra3xdh, 2024-10-19:
We are aware about this problem. It is related to bug in Qt itself. Wait for merging #995 for the solution.

ra3xdh, 2024-10-20:
Fixed by #995

closed 2024-10-20 by ra3xdh