ra3xdh/qucs_s #1007: Subcircuits node names are not made unique with ng-spice

Rmano, 2024-10-19:
Hi --- this is a quite complex thing to explain, so please bear with me...

The following project, opened with qucs 24.3.2 on a Manjaro system, with `ngspice-43`, behave in a very strange way. 
 
[strangesub_prj.zip](https://github.com/user-attachments/files/17446330/strangesub_prj.zip)

Notice that the project has the "LT" compatibility flag on.

I downloaded the INA826 library from the TI site; this is a very popular instrumentation amplifier. It works ok "as is" in LTspice (under Windows or Wine in Linux). If I use it directly in my circuit by picking it like this:

[image]

it does not work at all: my results are the following: 

[image]

As you can see, the amplifier is basically dead.

Now, I edited the text `.lib` file and changed all the occurrences of "GND" to "VSS" with a simple global search and replace, and to avoid clashes, I changed the name of the device to INA826E. And now...

[image]

It works flawlessly!!! So basically, the subcircuit node named "GND" is conflated with the external "GND" node name...

So this seems to be an encapsulation problem, and  I do not know if the problem is in `qucs-s` or in `spice.` Node names in the subcircuit should be private, don't they? At least, they behave in the LTspice app.

tomhajjar, 2024-10-19:
One of the schematics is missing the device.

[image]

Rmano, 2024-10-19:
You're right. Forgot to make save/all before creating the zip... I have re-loaded it above. Thanks and sorry for the problem...

tomhajjar, 2024-10-19:
I created a new project for you. It will also show you how to make sub-circuits and symbols.

There are v1 and v1.1 of the Spice model INA826 and you modified v1. I tested all 3 versions. I didn't dig into which model works under ngspice. The cleaned up project will make it easier for @ra3xdh to make a judgement.

The v1.1 model tests the same as v1.  I assume your modified v1e model works. 

The v1.1 model is supposed to be less "cluttered". It removed a number of unused sub-circuits. The pin order is totally different.

[image]
[image]
[image]

[OpAmp_INA826_prj.zip](https://github.com/user-attachments/files/17446897/OpAmp_INA826_prj.zip)

Rmano, 2024-10-19:
Very nice, thanks! Is there a tutorial somewhere on how did you do it? I tried but got stuck... even if I authored a simple tutorial for QUCS subcircuits (see attached), I got lost when trying to integrate the external libraries ... 

[ampli_qucs_tut.pdf](https://github.com/user-attachments/files/17446974/ampli_qucs_tut.pdf)

BTW, if this could be of interest for the documentation project, I'll be happy to contribute (adapting to qucs-s, that is, with also spice simulation).

tomhajjar, 2024-10-19:
Your tutorial is a start. 

It's easier for beginners to make sub-circuits from manufacturer Spice models if one looks at the many examples I posted in the Discussion area or examples included in the Qucs-S distribution. Finding Spice models compatible with ngspice is the most difficult thing. There is no guarantee a model that woks in LTspice or PSpice will work in ngspice. The ngspice guys can help determine if a model can be altered to work in ngspice or a modification to ngspice is justified.

I also posted Symbol examples that you can copy-paste into your subcircuit. I generally copy an existing symbol from the symbol examples into "symbols_blank.sch" and clean it up prior to placing it into the sub-circuit.

ra3xdh, 2024-10-26:
The Qucs-S doesn't apply any conversion to SPICE files attached via *Spice Library Device* and only checks if it contains .SUBCKT header. The specified file is included in .INCLUDE statement as is. It seems to be a problem of Ngspice. The GND may be a reserved keyword at some conditions. A minimal example reproducing this problem is required.

closed 2024-10-26 by ra3xdh

Rmano, 2024-10-26:
> The Qucs-S doesn't apply any conversion to SPICE files attached via _Spice Library Device_ and only checks if it contains .SUBCKT header. The specified file is included in .INCLUDE statement as is. [...]

Yes --- it seems that the GND symbol and name is global: https://www.reddit.com/r/AskElectronics/comments/sf6d66/ltspice_how_to_create_subcircuit_with_gnd_pin/ , so the bug is in the subcircuit specification (or maybe there is some strange version of SPICE where that doesn't happen).

Thanks