ra3xdh/qucs_s #1008: Verilog potentiometers not export to simulator correct equation for rotation

csrabak, 2024-10-19:
I'm trying to use the potentiometers and have the rotation parameter changed so to create a family of curves for the different settings of the cursors.

I stumbled a bit, and in the course of debugging I've arrived at this minimum example:

[image]

Each of the pots has the LEVEL parameter different, which from documentation (and a fine example of a Baxandall tone control) I understood the level 1 is linear taper, 2 is log and 3 'inv log'.

So, in the screen captured above, I would expect for the three pots at mid turn, the linear one giving half volt, the log one in the whereabouts of 10% to 20% of source voltage and the last one between 80% to 90% of the voltage.

However, all three show same 0.5V!

Inpecting the netlist exported (using Ngspice) yielded the following pertaining lines:

```
RPOT1_1 0 Vpot1 R='(10k)*(120)/(240.0)'
RPOT1_2 Vpot1 _net0 R='(10k)*(1.0-(120)/(240.0))'
RPOT2_1 0 Vpot2 R='(10k)*(120)/(240.0)'
RPOT2_2 Vpot2 _net0 R='(10k)*(1.0-(120)/(240.0))'
RPOT3_1 0 Vpot3 R='(10k)*(120)/(240.0)'
RPOT3_2 Vpot3 _net0 R='(10k)*(1.0-(120)/(240.0))'
```

It seems the implementation of either ignores the LEVEL parameter or the equations lead all to a linear function of cursor position.

ra3xdh, 2024-10-19:
The LEVEL parameter has no effect for SPICE mode.

ra3xdh, 2024-10-20:
Please wait until #959, #980, and #974 will be implemented. The rework of potentiometer is planned as a part of these issues. Closing this.

closed 2024-10-20 by ra3xdh