ra3xdh/qucs_s #1010: Sub-circuit made into Library doesn't work

tomhajjar, 2024-10-20:
In reference to #1007

I tried to add the INA826 to SpiceOpamp.lib. I made "INA826.sch" and confirmed it worked in "INA826_ac_tran.sch". I then made SpiceOpamp.lib with only the INA826 so it can be pasted into the "real" SpiceOpamp.lib. 

When I made "INA826_lib_ac_tran.sch" using INA826 from SpiceOpamp.lib, it doesn't work. As a test I created AD822.lib and it worked OK.

I'm confused...

On a side note, this could be used as an example how to add devices to existing Libraries...

[image]
[image]

[OpAmp_INA826_lib_prj.zip](https://github.com/user-attachments/files/17450705/OpAmp_INA826_lib_prj.zip)

Rmano, 2024-10-20:
The result seems correct, but what I pointed out in #1007 is that it should not needed to change the name of the `.lib` node from "GND" to "VSS"--- or am I wrong? I mean, the node names should be "insulated" by the external connections, no? 

I'll try to check this model. The warnings are a bit scary... they are not there in #1007 in the circuit that works.

ra3xdh, 2024-10-20:
> This could be used as an example/tutorial how to add devices to existing Libraries. 

This guide about libraries and subcircuits needs a translation: https://habr.com/ru/articles/805963/ See #554 Unfortunately nobody helps with rewrite the documentation. 

> When new library is made you must close/reopen Qucs-S to access it. Can this be resolved

I will look what could be done.

Rmano, 2024-10-20:
> This guide about libraries and subcircuits needs a translation: https://habr.com/ru/articles/805963/ See #554 Unfortunately nobody helps with rewrite the documentation.

I know zero of russian, but google can autotranslate the thing and I can volunteer (with a bit of patience, I have not a lot of time on my hand --- fortunately later circuitikz seems to be quite quiet) to adapt it. 

Is there a framework/style/master document to adhere to? I've looked at #554 but it seems a bit confused. Should the documentation be written in LaTeX (no problem, I'll be happy to do it).

ra3xdh, 2024-10-20:
The repository for documentation is here: https://github.com/ra3xdh/qucs_s_docs/ This link could be found in #554 There is no style guide. It uses standard `report` style. I have also created Overleaf project for Windows users: https://www.overleaf.com/project/65ed52fbd51e0623f11e285a So one can contribute ti the documentation without Github account.

Rmano, 2024-10-20:
I do not  see those warning... and it seems to work well with mixed supplies and saturations.

But with two INA826 I have this warning:

     Warning: redefinition of .subckt ina826, ignored

Ah, this should be called AmpInstr --- it's an instrumentation amplifier, not an OpAmp 😉

[image]

tomhajjar, 2024-10-20:
The warnings are for the subcircuit made using "SPICE netlist".

"Warning: redefinition"
ngspice doesn't like the spice file name to be the same as the internal device name. It doesn't cause problems. This goes away if you change either the file name or internal .SUBCKT name.

If you look at the attached library, it's called Instrumentation Amplifier

ra3xdh, 2024-10-26:
The library tree update after adding a new library implemented by #1027

closed 2024-10-26 by ra3xdh